ArduPilot/ardupilot #6410: Support for CANBus as battery data feed

ncx94, 2017-06-09:
#### Issue details
While looking for some king size batteries for our prototypes (12S, 50-80A)h, we found out that most of the big batteries don't use SMBus but CanBus for data-feedback. 
Because of that, i think it would be a great addition to be able to have canbus support for battery feedback.
I don't know how that could work/be integrated into UAVCAN, nor the differences between traditionnal CAN and UAVCAN, so it's just a free thought.

I'm trying to get into the code base and understand how to work on it/compile everything so i might give it a try in the next months but if anyone has better knowledge and would like to do it :)

Although generally only rovers use that much batteries, we could as well implement it to other firmwares.

#### Platform
[X] All

EShamaev, 2017-06-10:
I would not recommend mix of UAVCAN and other protocols on one physical bus.
So a separate protocol driver should be made.
Please share the protocol specification and then we can work out the options.

ncx94, 2017-06-12:
Yes obviously you wouldn't be using different protocols at the same time on the bus :)

As soon as I get a more detailled description of the exact protocol I'll share it!

EShamaev, 2017-07-10:
Any news on protocol?

ncx94, 2017-07-11:
Hi,

Sorry for the delay, I had a lot of things to deal with and I had forgottent to post here.

Attached is the descritption of the can bus part of the BMS used in the batteries..

Nick
[canbus-spec.pdf](https://github.com/ArduPilot/ardupilot/files/1138471/canbus-spec.pdf)

ncx94, 2017-08-02:
Hi @EShamaev,

did you have any time to look at the pdf?

thank you very much,

nick

EShamaev, 2017-11-12:
Sorry for long time to answer.
I think that adding SAEJ1939 as one more protocol is practical.
Please advice if the need is still actual.

ncx94, 2017-11-13:
Hi,

Sorry for the late answer!
Yes, the need is still actual and more than ever I would say!

That would be awesome if you could!

have a nice day,

nick

EShamaev, 2017-11-15:
May I ask if this is a commercial project?

ncx94, 2017-11-15:
It will be used on a commercial drone boat plateform yes but I think it can really benefit everyone that wants to go the smart battery way :)
I'm also trying to get the company to partner ardupilot but it's still in the works ;)

EShamaev, 2017-11-15:
I have evaluated the time it will take to implement SAEJ1939 as a separate driver, but as long as only one French device is present, which is rather "specific", custom and is not sold freely, I can not put too much effort into it.
However I can arrange to have it made if support is commercialized so the external person can be asked to do it under my supervision.

ncx94, 2017-11-15:
What do you mean by not sold freely?
If we need to pay for the feature, that's ok. Just give us an amount and we'll see how to integrate it :)

EShamaev, 2017-11-15:
I visited http://freemens.fr/en/ and could not find anything about FreeSafe Extended or any smart battery that can be purchased and no links to shops where it is sold.

ncx94, 2017-11-15:
Yeah ok I see what you mean. Actually freemens only makes the bms, then different batteries manufacturer use them in their products. The one we work with is Atomique batterie, also in France, who makes custom battery packs.

Anyway, if you can give us a quote on that job, that would be awesome :)

thank you very much,

nick

peterbarker, 2021-07-30:
We support uavcan battery monitors now.

closed 2021-07-30 by peterbarker