ArduPilot/ardupilot #6411: Build errors when building with HIL mode

sulania, 2017-06-09:
#### Issue details
I tried to build for HIL mode by uncommenting following line of code in file APM_config.h

`#define HIL_MODE              HIL_MODE_SENSORS    // build for hardware-in-the-loop simulation`

Following errors came

```
%% GCS_Mavlink.o
GCS_Mavlink.cpp: In member function 'virtual void GCS_MAVLINK_Copter::handleMessage(mavlink_message_t*)':
GCS_Mavlink.cpp:1817:9: error: 'gps' was not declared in this scope
         gps.setHIL(0, AP_GPS::GPS_OK_FIX_3D,
         ^
compilation terminated due to -Wfatal-errors.
make: *** [/tmp/ArduCopter.build/GCS_Mavlink.o] Error 1
```
Which i solved by adding copter object for follwing

#### Version
Master

#### Platform
[ X ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

OXINARF, 2017-06-09:
HIL is broken for a long time and not deemed as bug because the SITL system is much better for testing. There is an issue open for it, #508, so next time please search first before opening a new issue.

closed 2017-06-09 by OXINARF

magicrub, 2017-06-10:
That define  needs a huge comment explaining that.