ArduPilot/ardupilot #6413: Possible to move to latest PX4Flow firmware?

DonLakeFlyer, 2017-06-10:
@proficnc Had asked to me support flashing the firmware which ArduPilot uses for PX4Flow. Currently QGC by default uses this http://px4-travis.s3.amazonaws.com/Flow/master/px4flow.px4 to flash. The firmware which ArduPilot requires is older than this version.

I tried flashing ArduPilot Flow firmware (using custom flash capability) with QGC and running with that but it does not work due to the fact that it reports itself as a PX4 board. This confuses QGC and in turn causes it to not put up the PX4 Flow UI for doing things like focusing the lens. Newer versions of the firmware do not do this.

Is it possible to move ArduPilot forward to the latest PX4Flow firmware?

magicrub, 2017-06-11:
Hmm... I didn't know Ardupilot had a different version.

OXINARF, 2017-06-11:
As far as I know, the latest PX4Flow firmware still uses the old flow method, not the KLT one in the version ArduPilot wants to use - at least, there is an old PR for it open in there and I see no KLT mention in source code. Also, I'm not sure we have someone with the capability and time to test a new firmware that has gone through so many changes.

I don't think this needs to be discussed in the dev call, I think an answer from @priseborough is enough.

proficnc, 2017-06-11:
Mission planner supports our version...  concidering we have new hardware now, I can't see any benefit in changing the way all the existing users do this.  Surely there is a parameter or something that would indicate it's a flow unit?

priseborough, 2017-06-11:
Ardupilot changed to the KLT algorithm because back in 2014 it provided superior performance in low light and poor focus conditions, however as implemented in the px4flow sensor, the quality metric is too optimistic, so we rely on magnitude and innovation consistency checks.

Options as I see them are:

1) Update the firmware we currently use to provide the necessary identifiers: https://github.com/priseborough/px4flow/commits/klt_flow

2) Do a performance comparison to understand what we were gaining and giving away with using the latest PX4FLOW version. If the trade is favourable, patch the px4 version to use KLT and move Ardupilot to the new interface spec. This will drive change in mission planner and may require minor mods in the EKF to adjust quality metric thresholds.  

3) Do nothing - Mac users will have to run mission planner in a VM to make a PX4FLOW sensor Ardupilot compatible (or use command line tools)

rmackay9, 2017-06-13:
I think the action decided on the dev call was for me to attempt to pull Paul's branch into the ardupilot github area and then change the reported id from the flow sensor so the GCSs are happier.

DonLakeFlyer, 2017-06-13:
Yeah, that works as well. PX4Flow master reports 0 in HEARTBEAT.autopilot