ArduPilot/ardupilot #6414: Add Flight Mode: Calibration

marcmerlin, 2017-06-10:
As discussed on https://github.com/ArduPilot/ardupilot/issues/6244 , there should be a way to trigger recalibrations via RC.
The most common one is resetting the airspeed sensor to a value of 0 before takeoff with the airspeed sensor protected from wind. On most of my aircraft, I often see 10km/h or more in no wind conditions despite ARSPD_SKIP_CAL being left to 0 

@magicrub mentioned "The team is in favor of creating a new "Flight Mode: Calibration" which allows many different stick inputs to trigger any of the current calibrations (baro, compass, gyro, accel, airspeed ect ect). This mode can either be mapped to your RC transmitter like any other flight mode or you could boot up into it using the INITIAL_MODE param and any flight mode change (or the special exit stick combo) would exit the mode"

geofrancis, 2017-06-12:
i think the oled screen would be better for this, so when the machine is disarmed you do a stick combo to get into the menu like you would on an osd then you can have all the calibration options you want without having another flight mode.

marcmerlin, 2017-06-12:
@geofrancis that sounds nice, but leaves out the plenty of thousands of users who don't have the oled screen. 
Can you imagine what people would say if you told all copter users they now need an oled screen to arm their vehicle instead of stick down and to the right?
Sadly for plane users with an airspeed sensor, resetting an airspeed to 0 before takeoff is almost as essential, but cannot be done with any mobile GCS or with the sticks.
Sure, adding this to mobile GCS would be good, but still leaves out users without a telemetry radio or an mobile device with USB host adapter.
Being able to do this with stick movements would really be good.

Pedals2Paddles, 2017-06-12:
If it's a mode, it certainly opens up a lot of other possibilities.  For example, a stick gesture for parameter reset could be very useful on vehicles that have custom defaults (like a Solo).  A stick gesture that executes the preflight reboot shutdown would be great for post-calibrations.  Stick gestures are a lot easier than power cycling.

marcmerlin, 2017-07-03:
Since this flight mode is unlikely to happen anytime soon, and airspeed being misconfigured to be 10kph+ with no wind, this is starting to be a safety risk. One of my aircrafts stalled 3 times, once too close to the ground and was damaged heavily.
FBWA thought it was doing fine and the airspeed was 12kph over stall speed (gliding with motor off), but since airspeed values were wrong, it was right at stall speed, and it just stalled.
More details in https://github.com/ArduPilot/ardupilot/issues/6530
My suggestion is to add  a setting that allows airspeed to be reset to 0 when aircraft is armed.

jamie-wedderburn, 2020-05-07:
I'm a bit late to the game, but has there been any progress on a "calibrate" flight mode?

IamPete1, 2021-09-12:
I suspect the preference would now be to add a RC switch, https://github.com/ArduPilot/ardupilot/issues/12704 

Anyone strong attached to the flight mode idea?

IamPete1, 2021-10-01:
I guess no one is interested anymore?

closed 2021-10-01 by IamPete1