ArduPilot/ardupilot #6417: Default OreoLED flash frequency in range of cause epileptic seizures

hugheaves, 2017-06-11:
#### Issue details
I hadn't had chance to try out the default patterns with the new OreoLED / AP_Notify code, but as soon as I did, I was greeted by a very nice flickering LED pattern on startup. @Pedals2Paddles did a great job... it's almost hypnotic!

However, I realized that the frequency of the flash / flickering pattern (~7Hz) falls within the range (~3-30Hz) likely to cause epileptic seizures in people that have photosensitive epilepsy.  (Ref: https://www.epilepsysociety.org.uk/photosensitive-epilepsy)

As I don't suffer from this condition myself and am certainly not an expert, I can't weigh in on the true risk, but it might be prudent to change the default to a more relaxed frequency (2Hz ?). It would still be nice to be able to override the default to the faster version, as it looks great!

#### Version
Copter 3.5.0-rc8

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
3DR Solo

#### Hardware type
Pixhawk 2.0 Cube

#### Logs
N/A

Pedals2Paddles, 2017-06-11:
This must be an ArduPilot first... The super fast rate is defined as 150ms.  Which correct me if I'm wrong, means flashing 150ms on and 150ms off?  If my math is correct, would change the super fast rate to 350ms get it below 3hz?

Pedals2Paddles, 2017-06-11:
Also, for those unfamiliar with the Solo... What Hugh is referring to is the gyro initialization stage.  When the Solo is powered up, I have the LEDs start rapidly flashing in bright blue until the gyro calibration is complete. This signifies to the operator that the vehicle is calibrating and should not be moved.  The rapid flash rate is also used in non pilot controlled modes to indicate to the operator that they have been replaced :)

closed 2018-11-27 by Pedals2Paddles