ArduPilot/ardupilot #6420: High pitch angle in auto mode

Stijndbckr, 2017-06-12:
#### Issue details
Add the possibility to pitch more than 45 degree in AUTO mode, for the moment this is only possible in stabilize. 

#### Version
/

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quadrotor

#### Hardware type
Pixhawk

#### Logs
NA

OXINARF, 2017-06-12:
@rmackay9 Is this true? I thought that AUTO was just constrained by ANGLE_MAX.

Stijndbckr, 2017-06-12:
The reason for this is the limitation of the acceleration in the position controller:

 accel_total = norm(_accel_target.x, _accel_target.y);
    if (accel_total > accel_max && accel_total > 0.0f) {
        _accel_target.x = accel_max * _accel_target.x/accel_total;
        _accel_target.y = accel_max * _accel_target.y/accel_total;
        _limit.accel_xy = true;     // unused
     }

accel_max = gravitational constant. This means that the acceleration in x direction is limited with the gravitational constant and that the max pitch angle is 45°.

rmackay9, 2017-06-14:
Yes, @Stijndbckr is right.  AUTO is limited to 45 degrees.  It takes an extremely powerful vehicle to be able to keep moving forward at a 45 angle but they do exist.  Certainly we should only limit the lean angle by the vehicle's ANGLE_MAX parameter instead.

Stijndbckr, 2017-06-15:
@rmackay9 Do you think the attitude controller is able to handle this at the moment? Because Yaw will become roll at high pitch angles

lthall, 2018-01-11:
The attitude controller will handle a 90 degree pitch down just fine.

I will look at adding an additional angle max parameter for WPNav.

akselaase, 2021-06-19:
Are there any news on this issue? I've met the same limitation myself with a vehicle that could safely move at a 70 degree angle.

rmackay9, 2021-06-21:
@akselaase, 

Copter-4.1 allows leaning as far as 60deg in Auto mode.  Here is a screen shot from a simulator test.  In practice most vehicles cannot maintain their altitude with a lean angle of over 45 but if you do have a very powerful copter, then 4.1 can lean a bit more than 4.0.
[image]

peterbarker, 2021-07-30:
Closing as "done".

closed 2021-07-30 by peterbarker

rmackay9, 2021-07-30:
I think @lthall said that there's still something that needs doing.. 

4.1 definitely supports larger angles but there's something missing to allow it to lean to >60deg.

Muhammetyilmaz7, 2024-08-13:
How did you make this setting? How did you adjust the pitch setting while the drone was in autonomous mode? Can you explain step by step?