ArduPilot/ardupilot #6422: servo mixing in arduplane

geofrancis, 2017-06-12:
Now that arduplane has separate servo output and function parameters the logical next step would be to allow more than one function per servo similar to the mixer page on opentx.

so you could have a 
servo_1_mix parameter with value "A100E100" for 100% aileron 100% elevator etc
servo_2_mix parameter with value "A-100E100"
servo_3_mix parameter with value "T100"
and that would be your flying wing but with this you could have multiple functions per servo giving you things like crow mixing, speed brakes, split elevons for yaw etc without all the problems people are facing now when they try and build an exotic setup.

It would also make a lot of the current parameters redundant since you wouldn't need the parameters for elevon mixing, vtail, rudder/aileron mixing, flaperons, mix gain, to name a few since all that would be taken care of within the servo mix parameter.

955Ydfgg, 2017-06-24:
Great to learn that mixing can be more flexible using OpenTx. I want to accomplish something similar, but I need to stick to Arduplane Firmware modification to fit my application. What I essentially need is to vary the output to Four  RCn outputs, which currently only depend on Roll angle, (Aileron-type RCn_Function) as a function of Roll and Pitch angle (sth. like Mixing_Offset and Mixing_Gain for Elevons, but with those parameters different for each output, and functionally dependent on Roll and Pitch angle). Do you know whether somebody is already working on similar firmware modification?

tridge, 2017-10-30:
I do want to add mixing of this type, and I had planned to do it in a similar way to what @geofrancis describes. I haven't started on an implementation though. If someone else wants to start prototyping this then that would be great.

geofrancis, 2017-10-31:
I am willing to test anything anyone comes up with, I really think this along with #6903 could really transform ardupilot in a radical way, it could eliminate so many parameters, from wing and tail mixing, speed brakes, ic engine control, auto landing gear, failsafes, auto flaps, generator control, you wouldnt have to add more parameters every time somone wanted a feature since it could be all done with custom mixers and logic.

The best flight control software I have found in terms of airframe flexibility is openaeroVTOL, you can select how much of each input, gyro and auto level goes to each servo/motor channel so you can make it fly just about anything you can think of by changing the settings on the kk2 screen. so there is no need to make a frame type for every possible configuration that someone thinks of.

Ryanf55, 2019-01-12:
Hello, I'd like to help contribute in testing. I have a full house powered glider that uses advanced wing functions such as flaps, flaperson, crow, reflex, camber, and differential. I was tasked last year with replicating all these in Pixhawk and got most of the way there. I'd love to try any new mixers or help draft a good way to setup mixes. 
Thanks!

Ryanf55, 2019-01-31:
Is there any update on the new mixer or any plan been developed for implementation? It seems like a few ideas were bouncing around but nothing was quite decided.

bart989, 2020-10-24:
Isn't a LUA scripts a good option for this?

RFRIEDM, 2020-10-25:
No, Lua is not intended to control critical flight surfaces because the scripts are not safe from being terminated. I can't find a reference on this right now but do remember reading it.

bart989, 2020-11-03:
Well I've read it on the documentation on LUA scripts. Also not very happy with "low priorytet". But as I have tiltwing I didn't found a better option to make ailerons steer with yaw in vertical mode. How is model type specified? Maybe if you make a custom type, but I haven't found good instruction for this.

IamPete1, 2021-07-25:
The go to method for this is scripting, pcalls can make them more robust. You can read in the normal aileron, elevator ect output functions, mix them however you like and then output to the dedicated scripting functions.

closed 2021-12-08 by geofrancis