ArduPilot/ardupilot #6427: Accessing video frames before transmission for Bebop 2 drone using ardupilot

vishaldeyiiest, 2017-06-13:
#### Issue details
I'm working on the Bebop 2 drone and I wanted to try Ardupilot. But I see there are no libs for accessing video or camera modules for bebop 2. I see that ardupilot can access sensors like IMU, GPS. So using ardupilot, is it possible to access video frames before it goes to transceiver and then process frame bytes and then transmit it as usual?
#### Version
ardupilot project

#### Platform
Dev
#### Hardware type
Bebop 2

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-06-13:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

We'd love to have video working in the Bebop2. Unfortunately the handling of video is done in the same process as the Parrot autopilot (which can't be running at the same time as ArduPilot) so there isn't an easy way. I guess someone could develop a driver for it, but that is outside the scope of the ArduPilot autopilot firmware.

closed 2017-06-13 by OXINARF

rmackay9, 2017-06-13:
The answer in any case is that for the Bebop2, I don't think we have access to the camera.  We do on the Parrot Disco (plane) but the Bebop2 has a different API and I don't think we've ever been provided with the binaries to access the camera.