ArduPilot/ardupilot #6434: 2 classes with same name can cause compilation failure

night-ghost, 2017-06-15:
#### Issue details
there are 2 classes with the same name:

class OpticalFlow  in AP_OpticalFlow/OpticalFlow.h
and
class AP_HAL::OpticalFlow in AP_HAL/OpticalFlow.h
which are not relatives to one another

so if used "using namespace AP_HAL"  then compilation fails:

In file included from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_WayBack/AP_WayBack.h:57:0,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL_REVOMINI/HAL_REVOMINI_Class.cpp:30:
/mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_AHRS/AP_AHRS.h:140:28: error: reference to 'OpticalFlow' is ambiguous
     void set_optflow(const OpticalFlow *optflow) {
                            ^
/mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_AHRS/AP_AHRS.h:34:7: note: candidates are: class OpticalFlow
 class OpticalFlow;
       ^
In file included from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/HAL.h:15:0,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/AP_HAL_Main.h:19,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/AP_HAL.h:8,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL_REVOMINI/HAL_REVOMINI_Class.cpp:2:
/mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/OpticalFlow.h:17:15: note:                 class AP_HAL::OpticalFlow
 class AP_HAL::OpticalFlow {
               ^
In file included from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_WayBack/AP_WayBack.h:57:0,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL_REVOMINI/HAL_REVOMINI_Class.cpp:30:
/mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_AHRS/AP_AHRS.h:144:11: error: reference to 'OpticalFlow' is ambiguous
     const OpticalFlow* get_optflow() const {
           ^
/mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_AHRS/AP_AHRS.h:34:7: note: candidates are: class OpticalFlow
 class OpticalFlow;
       ^
In file included from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/HAL.h:15:0,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/AP_HAL_Main.h:19,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/AP_HAL.h:8,
                 from /mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL_REVOMINI/HAL_REVOMINI_Class.cpp:2:
/mnt/disk_d/src/quad-copter/ardu-copter/3.4.2-git/ardupilot.my.revomini.build/libraries/AP_HAL/OpticalFlow.h:17:15: note:                 class AP_HAL::OpticalFlow
 class AP_HAL::OpticalFlow {
               ^

#### Version
git

OXINARF, 2017-06-16:
> so if used "using namespace AP_HAL" then compilation fails:

Then don't use it? I'm not sure I understand if you are seeing this problem in master or in your modified code, can you clarify?

night-ghost, 2017-06-16:
Of course, I did so. I agree that I myself stepped on the rake, but you have carefully put this rake. There are lot of "use namespace AP_HAL" in the master too. 
Question is more theoretical - really two classes with the same name in one project is normal?

OXINARF, 2017-06-16:
OK, so the issue isn't in master. When you have such problem please either open a topic in the forum or talk about it in Gitter, we want to keep this issues list clean just for bugs in master or feature requests.

Having classes with the same name is perfectly normal in C++ projects, that's why namespaces exist - we should use them more, instead of prefixing all classes name with *AP_*.  I don't know why the optical flow one isn't prefixed, you can do a PR to change it if you want.
The *use namespace* isn't also a problem as long as the programmer knows what it is doing - the compiler should also catch all cases where it can have a problem, just like happened to you.

I'm closing this now as it isn't a problem in master, but feel free to answer.

closed 2017-06-16 by OXINARF

night-ghost, 2017-06-18:
Fine, issue closed again because "oh those russians". As you can see in error message, I was engaged in a library of the return by own track AP_WayBack, about which you seem to have asked. Now I see how much it  needed :)

tridge, 2017-06-19:
hi @night-ghost, I wouldn't object to a PR that renamed the HAL OpticalFlow class to something else if you can think of a good name (maybe OptFlow ?)
Apart from the case you mention, I can imagine it helping for users using programmers editors that find class by name (eg. etags in emacs)

night-ghost, 2017-06-20:
Any name will be good if it do not confuse the code as it is now. 

> programmers editors that find class by name

or Eclipse as I use :)

PS. AP_WayBack library now compiles in ArduCopter and works in debug (via test GUI and serial), I find some points when it should be called but I can't find a point in which it should inject breadcrumb instead of Home point. So now I can't do a PR with fully working code.