ArduPilot/ardupilot #6436: DGPS is not injected in Plane

antonvh, 2017-06-15:
#### Issue details
My hardware setup is a RPi3 with a Navio2 and a reach RTK module on /dev/ttyAMA0. 
When I run ArduRover (latest) on that setup DGPS works fine. If on the exact same setup I run ArduPlane there is no DGPS injection anymore.

#### Version
ArduPlane 3.7.1

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Quadplane

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Navio2

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2017-06-15:
Plane v3.7.1 is actually getting quite old. Please try v3.8beta and report back.

antonvh, 2017-06-15:
3.8.0 works. I didn't dare to fly with it, as it wasn't labeled stable.

antonvh, 2017-06-23:
3.8.0 has a host of other issues. How can I fix this in 3.7.1? Which commit/pull request fixes dgps?

OXINARF, 2017-06-23:
Many things changed since then, there isn't a single commit or pull request I can link to. In any case, this is fixed, there is no reason to keep this open.

Also, if you have issues with 3.8 please post them in the forum, that's the only way they can be fixed.

closed 2017-06-23 by OXINARF

WickedShell, 2017-06-23:
I think I mentioned this somewhere already, plane 3.8 supports a new
message in addition to the old one for injecting data, I suspect your GCS
is only sending the new message that the old version on plane doesn't
understand