ArduPilot/ardupilot #6441: Problem with LittleBee 30A Opto

gppsoft, 2017-06-15:
#### Issue details
When the throttle reaches a minimum at the zero position of the stick, the revolutions increase, and then normalize. Engine control is delayed. How to fix this?

https://www.youtube.com/watch?v=YFfkhGe2xoI

#### Version
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[x] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Quadrocopter X

#### Hardware type
Pixhawk + LittleBee 30A Opto(BlHeli 14) + OneShot125

#### Logs
_Please provide a link to any relevant logs that show the issue_

rmackay9, 2017-06-16:
Support questions should really go into the discuss forums: http://discuss.ardupilot.org/c/arducopter/copter34 so I'm going to close this issue.  Sorry to be strict but if we're not, then our issues list (for confirmed bugs and enhancements) just turns into a support forum.

I think maybe what you're seeing is normal.  There's some delay in response during the spin-up of the motors and some delay as the vehicle determines if it's landed or not.  Maybe just trying flying it and see if it's ok?

closed 2017-06-16 by rmackay9

gppsoft, 2017-06-16:
OK. Thank you. I'll try to fly