ArduPilot/ardupilot #6452: AC3.5rc8 Need SF40C driver upgrade

roque-canales, 2017-06-17:
Using SF40C if bad readings occur, SF40C return ZERO values,.....

This induce wooble and jearky  drone when using avoidance in althold, wich is very uncomfortable and make unusable avoidance feature.

It seems that filter is incorporated in loiter mode. So loiter mode avoidance with SF40C is usable.

Idea will be to filter this bad readings ZERO values that not match reality.

You will find attached log, and between log line 20K and 40K you can see bad readings zero value and non zerovalue that mach reality.
[74 01-01-2000 01-24-46.zip](https://github.com/ArduPilot/ardupilot/files/1082671/74.01-01-2000.01-24-46.zip)

OXINARF, 2017-06-19:
@rmackay9 Any input on this?

rmackay9, 2017-07-06:
Yes, we should make this fix.  I didn't see this issue so I created another one.  I'm going to close mine ('cuz earlier one is the one we normally keep).
https://github.com/ArduPilot/ardupilot/issues/6539

rmackay9, 2017-07-08:
I've added a fix to master and this should go out with AC3.5.1.
https://github.com/ArduPilot/ardupilot/commit/d636e85eac4fb81081fad7f38ca3cffc2730fcee

Thanks again for the report.

closed 2017-07-08 by rmackay9

roque-canales, 2017-07-12:
I think that it must be better to add as for the RNGFND, min and max value filtered by for example:
PRX_READ_MIN
PRX_READ_MAX.

Like this each one can set it as it wants for improve integration.