ArduPilot/ardupilot #6457: AC3.5 Lightware I2C lidar never reports unhealthy

rmackay9, 2017-06-19:
If a Lightware LW20/I2C (and probably the SF10/SF11 using I2C) is connected according to the wiki http://ardupilot.org/copter/docs/common-lightware-lw20-lidar.html it seems that the lidar will never be reported as unhealthy even if the sensor is detached (or not present).

A lightware LW20/Ser (i.e. using serial interface) does report itself and unhealthy if the sensor is detached.

rmackay9, 2017-07-06:
I've double checked this and it seems to correctly report that it's unhealthy if the I2C cable is disconnected.

closed 2017-07-06 by rmackay9

muramura, 2017-07-06:
@rmackay9 san.  If you need to investigate in LW20 of I2C communication, we will investigate if you can receive a message.

rmackay9, 2017-07-06:
@muramura, great, thanks.  I thought I say it not working a few weeks ago but I've looked at the code and justed tested it and it seems fine to me..