ArduPilot/ardupilot #6460: Copter does not distinguish between LAND_SPEED and LAND_SPEED_HIGH paramter

GekoCH, 2017-06-19:
#### Issue details
I currently use an IRLock for the landing. Now since the landing speed was really slow I decided to play with the LAND_SPEED and LAND_SPEED_HIGH parameter.

I changed the LAND_SPEED to 75cm/s and the LAND_SPEED_HIGH the same as WPNAV_SPEED_DN to 300cm/s

However the Copter still descended for the whole time with 75cm/s . I then changed the LAND_SPEED to 300cm/s and the copter did descend with 3m/s for the whole descend (was a bit a rough landing).

So the copter definitely does not distinguish between LAND_SPEED and LAND_SPEED_HIGH 

The landing command was sent through the MP. 

#### Version
v3.4.6

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
X-Copter

#### Hardware type
PIXHAWK

rmackay9, 2017-06-19:
Thanks for the report.  this is probably an issue only when doing precision landing so I've modified the title.  No less valid, just clarifying it.

OXINARF, 2017-06-19:
@jschall Since you have done some changes to this part of the code about a year ago, can you give your input here?

jschall, 2017-06-20:
This is not the expected behavior. Can we get a log of this?

Note that the vehicle will always descend at LAND_SPEED below 10m.

GekoCH, 2017-06-20:
you say 10m hmm I found some parts in the code where it should switch to land_speed below 2m in the control_land.cpp part. Where in the code is this located where it switches at 10m?

logfiles:
http://www.quadcam.ch/data/log01.bin
http://www.quadcam.ch/data/log02.bin

Andy

jschall, 2017-06-20:
LAND_START_ALT is hard-coded to 10m.

jschall, 2017-06-20:
Looks like it most likely has to do with your rangefinder and/or rangefinder config. Your vehicle always thinks it is at 6m AGL when it is up and away.

GekoCH, 2017-06-20:
ok thx for the hint I will check this!

GekoCH, 2017-06-20:
ok rangefinder works normally I got an XL-MaxSonar®-EZ4™ with a max range of 765 cm.
The parameter is set to a max range of 700cm don't know why it stuck at 6m...

GekoCH, 2017-06-20:
ok did a new flight with adjusted RNGFND_MAX_CM to 500 and now the copter distinguishes between  LAND_SPEED and LAND_SPEED_HIGH.

However the PLND does only correct right above the ground at around 2m. The copter slowly descends from 10m towards ground and suddenly at around 2m altitude it starts to correct towards the IRLock why not earlier?

OXINARF, 2017-06-22:
Looks like the reported issue was a configuration problem and not a bug, so I'm closing.

@GekoCH There is nothing in code that limits horizontal correction before reaching 2m HAGL. It will slow down at that height if doing precision landing which is probably why you can see it better.

closed 2017-06-22 by OXINARF