ArduPilot/ardupilot #6477: How to understand cantFuseFlowData in EstimateTerrainOffset() in EKF3? [solved]

luweikxy, 2017-06-21:
Hi, there is an anther problem that has puzzeled me for several days. If you are available, would you please take a look?

in NavEKF3_core::EstimateTerrainOffset(), the peice of code to check the consistency of innovation:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_NavEKF3/AP_NavEKF3_OptFlowFusion.cpp#L99

    // don't update terrain offset state if there is no range finder
    // don't update terrain state if not generating enough LOS rate, or without GPS, as it is poorly observable
    // don't update terrain state if we are using it as a height reference in the main filter
    bool cantFuseFlowData = (gpsNotAvailable || PV_AidingMode == AID_RELATIVE || velHorizSq < 25.0f || losRateSq < 0.01f);

Here you said don't update terrain state if without GPS. But you konw gps is not available in door where optical flow usually is often used. Unit which can measure height is not only GPS but also baro and so on.So if GPS is not available and baro is available, in selectHeightForFusion(), activeHgtSource will change to HGT_SOURCE_BARO, the height still can be measured by baro, So why we cann't fuse flow data?

That is to say, baro also can measure height like GPS, so althought without GPS, if baro exists, is it still possible to update terrain state?
Many thanks!

luweikxy, 2017-06-21:
Paul has replied this question in gitter, now I copy to here:

============================================================

Paul Riseborough @priseborough

With this simple 1-state terrain offset estimator, you can't use optical flow data to estimate height if you don't have an independent of earth relative velocity. You can use optical flow and height to estimate velocity or velocity and flow to estimate height, but not both. In theory height would be observable if we built the terrain offset state into the main filter and performed sufficient velocity changes.

============================================================

My unedrstanding after Paul's instruction:

If gps is not avalible or aidding mode is relative which means we donnt want to use GPS bacause the GPS data is bad althought GPS data exits, absolute velocity cannt be got, so we cant use optcal flow to get ground height because we cannot get the predicted body angular rate.

namely, when cantFuseFlowData is true, it means that there is no GPS or we donnt want to use GPS, so the absolute velocity of vehicle cannt be got, and the optical flow cannt be trusted, so we cant fuse flow data.

Many thanks to @priseborough

OXINARF, 2017-06-22:
@luweikxy I'll try to explain again by points:

- Questions: use Gitter
- Documenting questions and their answers (like you did here): use the forum
- Bugs that you are 100% sure about or feature requests: use this issues list
- Contributing code, whether to fix a bug or a new feature: open a pull request

closed 2017-06-22 by OXINARF