ArduPilot/ardupilot #6486: APM 3.5 Beta - Copter Crash

josepgomes, 2017-06-21:
#### Issue details
Everytime we pilot for a while the copter crashs because the motors simply stop.  

#### Version
3.5 beta

#### Platform
[   ] All
[   ] AntennaTracker
[ x ] Copter
[   ] Plane
[   ] Rover

#### Airframe type
Quadcopter

#### Hardware type
Erlebrain3 from Erle Robotics

#### Videos

https://youtu.be/edjqyYt1_Ew?t=25s

https://youtu.be/iQi6sdzv5eg?t=1m20s

Thank you for your time!

Pedals2Paddles, 2017-06-21:
Can you download and post dataflash logs?  Is it always after a consistent amount of time, or seemingly random?  I hate to ask the obvious, but you're not running out of battery power, are you?

khancyr, 2017-06-21:
Ok known problem on erle brain : see  http://discuss.ardupilot.org/t/copter-3-5-0-rc7-unexpected-crash-on-erlebrain/18465/10

LanderU, 2017-07-04:
We update the out OS version, the problem disappear with this version: http://forum.erlerobotics.com/t/update-frambuesa-os-for-erle-brain-3/3057

OXINARF, 2017-07-04:
Thanks @LanderU, closing this then.

closed 2017-07-04 by OXINARF