ArduPilot/ardupilot #6489: Quadplane disarm does not work. (Min throttle, trim). plane-VTOL

antonvh, 2017-06-22:
#### Issue details
I calibrated the Quadplane ESCs first to go from 980 to 2000.
If I check the calibration by setting Q_ESC_CAL to 1 and mode to QSTABILIZE I can confirm the calibration is fine. I also adjusted both SERVx_TRIM and RCx_TRIM on all motor channels to 980.
However: if I disarm the plane the motors get a PWM of 1100 and start spinning.

This was totally weirding me out, as servoraw over mavlink said the channels were outputting 980. But measurement showed that it was in fact 1100.

Switching back to 3.7.1 removed the problem completely.

I think it might have to do with the new SERVOx paramaters that are not well implemented.

Sadly 3.7.1 is no option for me, as it has a GPS Inject bug that is fixed again in 3.8.0...
#### Version
3.8.0b

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QUADPLANE wing

#### Hardware type
Navio2/Rpi

#### Logs
_Please provide a link to any relevant logs that show the issue_

guiseco, 2017-06-22:
Just for checking:

ARMING_ Parameters
ARMING_RUDDER: Rudder Arming
Note: This parameter is for advanced users

Control arm/disarm by rudder input. When enabled arming is done with right rudder, disarming with left rudder. Rudder arming only works in manual throttle modes with throttle at zero +- deadzone (RCx_DZ)
Values
Value 	Meaning
0 	Disabled
1 	ArmingOnly
2 	ArmOrDisarm

antonvh, 2017-06-23:
I was arming/disarming through the GCS, not with the rudder.

magicrub, 2017-06-23:
This sounds serious. Please attach looks so we can look into it.

antonvh, 2017-06-23:
OK. Will do so monday.

magicrub, 2017-06-23:
need .tlog, want .bin

antonvh, 2017-06-27:
Her you go. https://www.dropbox.com/s/dn2r002fw11xqg7/00000018.BIN?dl=0

The quadplane motors are on RC5-8. You can see that they change PWM, even when disarmed!

I didn't save a .tlog

tridge, 2017-07-14:
hi @antonvh, I tried to reproduce this with a navio+ (I don't have a navio2). I set it up with the parameters from your log file, then armed and disarmed it. My logic analyser shows the expected 960 PWM width when disarmed.
ahh, now I see the cause. You had Q_ESC_CAL=1. That enables pass-through from the throttle to the motors, and is meant for doing ESC calibration. So this is exactly the expected behaviour. If you had rebooted it would have automatically set Q_ESC_CAL back to 0.
I'll close as its not a bug.

closed 2017-07-14 by tridge

antonvh, 2017-08-02:
Today I collected more log files to prove it is a serious bug.

https://www.dropbox.com/s/zzx0rz7vxymtnyw/00000026.BIN?dl=0
Log 00000026.BIN is the most interesting. It's the log where I go through the whole ESC calibration procedure. 
At around 250000 I'm putting power on the ESCs with full throttle. Some time later I set throttle to 0 and than play around with the throttle to verify the motors react to the stick.
At line 511179 I'm setting Q_ESC_CAL back to 0, for normal operation. 
Now the funny things start to happen.
And then at 563374 I disarm throttle. Now what is not visible in the logs is that the quadcopters motors started spinning at 1500 pwm! I'll make a movie of this later.
I then disarmed and the motors started slowly spinning at 1065 pwm, which is weird considering the configured TRIMs and MINs.
Not cool.
At 594157 I started crying and stopped arduplane 3.8.0b5.

antonvh, 2017-08-02:
For your entertainment two more videos showing the bug
https://www.youtube.com/watch?v=jU24Cs-AEF4&feature=youtu.be
https://www.youtube.com/watch?v=p7VhrdnM54k&feature=youtu.be