ArduPilot/ardupilot #6491: cam messages repport last POS recorded and not EKF Instant POS

roque-canales, 2017-06-22:
#### Issue details
_Please describe the problem, or desired feature_

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_

#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-06-22:
What is the issue here? There is no description and from the title I don't see any problem. I assume you are talking about *current_loc* which is updated in every loop, so there isn't any more real-time than that.

WickedShell, 2017-06-22:
MAVLink side is incorrect due to the deferred stuff, but given that it's talking about CAM messages I'm assuming the issue is about DF logs...

OXINARF, 2017-06-22:
@WickedShell Yes, message can be incorrect if it is deferred as well as it can be dropped entirely, so relying on it isn't a great idea. But, as you said, this looks to be about DF logs, so I don't know what the issue is.

closed 2017-06-23 by roque-canales