ArduPilot/ardupilot #6498: Request to include GPS heading and GPS speed in native Frsky passthrough protocol

rsilk1949, 2017-06-24:
#### Is_Please describe the problem, or desired feature_
A few telemetry settings are not available in native Frsky passthrough, but are available in the other protocols.
Would like to see GPS heading and GPS speed in particular.

#### Version
_What version was the issue encountered with_
3.4

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
PixHawk with C&T telemetry cable.

#### Logs
_Please provide a link to any relevant logs that show the issue_

floaledm, 2017-06-28:
@rsilk1949 Thanks for your interest. GPS heading and speed were omitted by design to save on the very limited (1,200bps) bandwidth of the FrSky link. Instead, ground speed (using GPS/INS filter) and yaw are transmitted using native Frsky passthrough.

Could you explain your specific need for those values?

rsilk1949, 2017-06-30:
Can you pls provide a link to a definitive list, or an include file, that shows  all the IDs that are output in passthrough (protocol 10) ? There must be one somewhere :)

magicrub, 2017-06-30:
Have you checked the wiki?
http://ardupilot.org/plane/docs/common-frsky-telemetry.html?highlight=frsky

Do we need to add IDs to the packets in the chart?

floaledm, 2017-06-30:
@rsilk1949 The wiki contains all the information you're looking for, including the IDs in a speadsheet that the ersky9x dev used to parse the passthrough protocol: http://ardupilot.org/plane/docs/common-frsky-passthrough.html?highlight=spreadsheet

Thanks Tom. I think it would be too much for this chart which is about choosing which protocol to use by knowing what information is available with each protocol.

rsilk1949, 2017-06-30:
OK I checked the spreadsheet and can figure out nearly all the settings mentioned in the table.
However, there are some I think may be missing. If so, can someone update the spreadsheet please?:

a) Flight timer (automatically counting while in flight)
b) Battery pack capacity remaining (%)

Is Ground speed (using GPS/INS filter) = VELANDYAW Horizontal velocity?
Is YELANDYAW/Yaw = current compass heading?

floaledm, 2017-06-30:
@rsilk1949 The spreadsheet is accurate and describes what is transmitted using the passthrough protocol. The table comparing the protocols shows what can be obtained either directly or indirectly (through conversions/calculations).

> Is Ground speed (using GPS/INS filter) = VELANDYAW Horizontal velocity?
Yes, except for plane (if airspeed is available, airspeed is transmitted instead).

> Is YELANDYAW/Yaw = current compass heading?
It's not from the compass heading directly but the yaw from the AHRS (yaw_sensor).

For more details, you can refer to the code to find out exactly which ArduPilot variables are transmitted.

rsilk1949, 2017-06-30:
So does that mean that battery % remaining is not sent directly but needs to be be calculated using
(battery capacity - mah consumed)*100/Battery capacity
?

floaledm, 2017-06-30:
Yes, the same way ArduPilot calculates it. Transmitting batt% would therefore not add info that is otherwise missing.

rsilk1949, 2017-07-01:
Thanks. I will close this issue then since I can use the spreadsheet together with the table to get what I need.

closed 2017-07-01 by rsilk1949