ArduPilot/ardupilot #6499: distance_sensor Problem

Ahrovan, 2017-06-25:
I send range finder via MAVLINK. I set below setting:
1)[image]
2) 
[image]
3) wiring done and value send by this code with ~10Hz loop:
```
	  dis_msg.radiation_type = 0;
	  dis_msg.field_of_view = 25;
	  dis_msg.min_range = 1;
	  dis_msg.max_range = 40;
	  dis_msg.range = Lidarvalue.data;

	  dis.publish(dis_msg);

```
4) show sonar range Right
[image]
5) but this error occur and can not arm copter ( sometime say "error pos vert variance" and crush ):
```
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
PreArm: check range finder
EKF2 IMU1 is using optical flow
EKF2 IMU0 is using optical flow
PreArm: check range finder
PreArm: Need 3D Fix
EKF2 IMU1 tilt alignment complete
EKF2 IMU0 tilt alignment complete
EKF2 IMU1 initial yaw alignment complete
EKF2 IMU0 initial yaw alignment complete
Initialising APM
Initialising APM
```
[Copter-3.5.0-rc8- pixhawk - Mavros 0.19 + KINETIC ROS]

Ahrovan, 2017-06-25:
I check this note:
```
Copter 3.3-rc2/rc3 02-May-2015
Changes from 3.3-rc1
...
2) Optical flow improvements:
    a) Range Finder pre-arm check - lift vehicle between 50cm ~ 2m before arming.  Can be disabled by setting ARMING_CHECK to "Skip Params/Sonar"
    b) Vehicle altitude limited to range finder altitude when optical flow is enabled 
```
my range  finder limit is 5 ~ 1000 cm  (after Modify). and lift vehicle between 50cm ~ 2m but mission planner say again:
```
PreArm: check range finder
EKF primary changed:1

EKF primary changed:0

PreArm: check range finder
EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

PreArm: check range finder
EKF primary changed:0

EKF primary changed:1

PreArm: check range finder
```
[image]

Ahrovan, 2017-06-26:
When lift vehicle ~20cm above ground i see this ressult:
[image]
```
EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

EKF primary changed:0

EKF primary changed:1

EKF primary changed:0

EKF primary changed:1
```

OXINARF, 2017-06-26:
This issues list is for feature requests and confirmed bugs. For user support please post in the forum (http://discuss.ardupilot.org).

Check that you are sending MAVLink messages fast enough - you need to have less than 500ms interval.

closed 2017-06-26 by OXINARF