ArduPilot/ardupilot #6510: 3.8rc5: Auto take off pitched down and sideways to crash

marcmerlin, 2017-06-27:
I'm not making sense out of this. I've used auto takeoff on that aircraft plenty of times, no issues.
Before launch, I made sure the servos were going in the right direction when in auto, and they seemed ok before I launched.
Aircraft pitched down into the right and crashed on its nose at full power. Worst, power was still running after the crash because I had to disable the crash detection since it was cutting power erroneously on auto takeoffs (reported last year). 

https://youtu.be/sqsABITE530?t=46s

MAV> graph ATT.Roll ATT.Pitch AHR2.Roll AHR2.Pitch GPS.Spd:2 ARSP.Airspeed:2
[image]

Questions: 
1) is there an easy way to graph magnetic and GPS derived flight heading? (haven't found this)
2) this kind of full power crash is kind of damaging. Any idea what else I should do before auto takeoff to avoid such a problem in the future?

marcmerlin, 2017-06-27:
[00000037_bix2_auto_takeoff_crash.BIN.gz](https://github.com/ArduPilot/ardupilot/files/1106095/00000037_bix2_auto_takeoff_crash.BIN.gz)

@magicrub would you mind having a quick look to see if it's a 3.8 bug, or some other issue that caused this crash?

marcmerlin, 2017-06-30:
This may help debug https://github.com/ArduPilot/ardupilot/issues/6522 further, but obviously I can't look at this auto takeoff problem anymore now that the plane has become a pile of foam bits.
I thankfully do have auto takeoff still working ok on my 2 other aircraft (bigger and more expensive), but I'm fairly nervous now...

magicrub, 2017-07-01:
I'll take a look. Sorry for your loss. We very much appreciate your testing.

tridge, 2017-07-12:
@marcmerlin do you happen to have a previously successful flight log from the same aircraft? Ideally an auto-takeoff, but even without would do. I'd like to compare servo outputs

tridge, 2017-07-12:
@marcmerlin the RCOU log shows the ailerons were hard over left, and elevator fully up during the takeoff, which makes sense as the aircraft was rolling right and pitching down. On a bix2 the airspeed you had should have been enough for it to react. Everything else looks fine, it just looks like the ailerons and elevator weren't effective. If your preflights were right and the elevator and ailerons were moving in the right directions then it had no reason to crash.

tridge, 2017-07-12:
> is there an easy way to graph magnetic and GPS derived flight heading? (haven't found this)

I use this graph:
graph GPS.GCrs ATT.Yaw mag_heading_df(MAG,ATT)
mag_heading_df() needs some extra functions that I put in my ~/.pymavlink/mavextra.py
I've put a copy here:
http://uav.tridgell.net/tmp/mavextra.py

> this kind of full power crash is kind of damaging. Any idea what else I should do before auto takeoff to avoid such a problem in the future?

if flying a setup with lots of changes (eg. major firmware revision) for first time then I'd suggest a FBWA takeoff to start with. That can still give a lot of damage with a reversed control surface, but at least you can reduce power.

If you are really nervous then a run and throw launch in FBWA without the motor on can be a worthwhile check. That allows you to check stabilisation and CoG with minimal energy.

>  I'm just wondering whether the GPS module being a bit loose could have affected the plane's view of level flight although in this case it seems that the mismatch was pretty severe anyway.

it didn't affect this flight, as all 3 attitude estimates were consistent (DCM and two EKFs). The compass can affect attitude, but will rarely cause a problem on fixed wing as long as you have GPS lock.

marcmerlin, 2017-07-12:
thanks for having had a look at this one. I think we can safely roll it into  https://github.com/ArduPilot/ardupilot/issues/6522 and continue there.

closed 2017-07-12 by marcmerlin