ArduPilot/ardupilot #6511: Boat Mode INS_GYR_CAL

andyboatdiver, 2017-06-27:
#### Issue details
_Please describe the problem, or desired feature_

Iam currently testing Boat Mode with Coptor 3.5.0 - RC 8 with green cube on Solo using Solex. I note that when setting INS_GYR_CAL to 0 via Tower and rebooting, the LED sequence stops at the Blue disco lights and does not complete the initialization to Red/Green White/Yellow. Solex will not control the LED's and is not in control. Using Tower, that will arm Solo and the solo controller will allow flight in manually or stabilize.

Has anyone really managed to make boat mode work with Solo,green cube and 3.5.0 - RC 8? If so how?

I have a boat ready for testing and want to see how EKF2 will deal with movement on open seas. I have perfected hand catching without any finger loss.

#### Version
_What version was the issue encountered with_

Copter 3.5 RC 8

#### Platform
[  ] All
[  ] AntennaTracker
[ X ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Solo 

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Solo with 2.1 Green Cube  running Tower and Solex

#### Logs
_Please provide a link to any relevant logs that show the issue_

magicrub, 2017-06-27:
>>I have perfected hand catching without any finger loss

Congrats...So far....

andyboatdiver, 2017-06-27:
Have not touched the blades yet, but wear a Kevlar glove, just in case.

rrr6399, 2017-06-28:
I'd wear a fencing suit with a Kevlar neck cover as well! :-) 

Although, this girl was unafraid:
https://youtu.be/XNm9aomNhRw?t=532

rmackay9, 2017-06-28:
I've tested setting INS_GYRO_CAL to 0 on a regular Cube (aka Pixhawk2) and it seems fine...
For the most part BoatMode should just work.  I'm sure you've already seen the wiki page but I'll include it for others as a reference.  http://ardupilot.org/copter/docs/boat-mode.html

Reallly support issues should go into the forums so I'm going to close this if that's OK.  http://discuss.ardupilot.org/c/arducopter/copter-3-5

andyboatdiver, 2017-06-28:
It appears to be a solo specific problem.

"
The LEDs staying in blue strobe under those circumstances is something I
need to fix. The LEDs are waiting for the gyro calibration to begin and
then complete. With the gyro Cal disabled, it never happens and they never
switch out of blue strobe. Despite that, the solo is operating normally.
It's just the LEDs that won't do their thing."

Pedals2Paddles, 2017-06-28:
I'm going to reopen this issue and assign it to myself. It's a legit bug in the OreoLED file that I need to fix.  It is only the LEDs, and only on a Solo.  I don't know if I'll get this handled by RC9 or 3.5.0.  So worst case is it gets fixed in 3.5.1,

Pedals2Paddles, 2017-06-30:
I've opened a PR (https://github.com/ArduPilot/ardupilot/pull/6524) to rectify this.

Pedals2Paddles, 2017-09-01:
Closed. Fixed with https://github.com/ArduPilot/ardupilot/pull/6524 in ArduCopter 3.5.2.

closed 2017-09-01 by Pedals2Paddles