ArduPilot/ardupilot #6520: BATT2 monitor voltage clipping

Georacer, 2017-06-30:
#### Issue details

##### Background:

I was trying to interface some analog Angle-of-Attack and Angle-of-Sideslip sensors with Pixhawk 1 to get some logging done. Nothing fancy, just proof of concept.
I was trying to find a way to introduce those max3V3 readings into the Pixhawk 1 so I decided to connect the two signals on the 3V3 ADC and read them as a 2nd battery monitor, in lack of a better alternative.

I set up the voltage scaling for both pins to 90degrees/3.3V = 27.27. After doing that, I noticed the following in *Mission Planners telemetry graph*:
* The current2 reading would correctly go from 0 to 90, for the full input voltage range
* **The voltage2 reading would climb up to about 65 and then fold over to 0**

Since this was too close to 64k, I parsed the battery_monitor code in search for some uint32_t declaration, but the voltage is declared as float, so I don't know what to make of this.

In short, steps to replicate:
* Input 3V3 at pin 13 or 14
* Set this pin as BATT2 analog voltage
* Set scaling factor to 27.27
* Watch telemetry in Mission Planner not reach a value of 90

I don't have a tlog at hand right now, I'll power up the plane later in the day to get one. Just wanted to type this before I got distracted.

#### Version
ArduPlane 3.7.1

#### Platform
[X] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
N/A

#### Hardware type
Pixhawk 1

#### Logs
Available upon request

OXINARF, 2017-07-03:
Still waiting on tlog. Have you confirmed it isn't a MP issue?

Georacer, 2017-07-03:
Right, so, I got some logs on the issue.

The .bin file does not look like it displays the issue.
When I replay the tlog file on Mission Planner, I see the issue. I don't know of any other tlog viewer except MP.
[logs.zip](https://github.com/ArduPilot/ardupilot/files/1119864/logs.zip)

Georacer, 2017-07-03:
I know the codebase of Mission Planner even less than ardupilot's, so I can't confirm or deny that it is a MP issue.

OXINARF, 2017-07-03:
This is a MAVLink limitation. You looked at the code in ArduPilot but didn't check the MAVLink message where voltage is sent as a uint16, so it can only go to 65535 mV.

I'm closing this as there is nothing we can do about it, unless new MAVLink messages are added - and that doesn't look very useful at this point, current limitation should be enough for most people.

Next time, please post in the forum, this list should only be used for confirmed bugs.

closed 2017-07-03 by OXINARF

Georacer, 2017-07-03:
... and the only reason that the value of the current is not affected is that the unit of measurement is 10*mA (based on the HEARTBEAT message). Ok. This is quite random, but it's the reason why this doesn't work and I understand that it can't be changed easily.
Thanks for the pointer.