ArduPilot/ardupilot #6523: AP 3.8rc5: after landing FBWA and AUTO twitched aileron servos left and right

marcmerlin, 2017-06-30:
I had a proper uneventful flight with AUTO takeoff and FBWA working fine.
After landing (look for MANUAL 765sec), the aircraft was on the ground, I fixed some video issues and I was going to go for another flight.
When I put FBWA the aileron servos were going up and down like crazy.
I then tried AUTO and same thing happened.
Clearly I didn't launch the aircraft given the situation, the horizon in the OSD looked good, I have no idea what the aircraft was trying to do, but typically it does this when it thinks it's upside down or at a high bank angle and trying to right itself back up.
I was worried that I'd get to experience https://github.com/ArduPilot/ardupilot/issues/6522 again.

I however power cycled the aircraft, and after boot and init FBWA and AUTO looked fine, and the aircraft launched ok.
Still, this makes me quite worried that it could have been a serious bug I just worked around by rebooting.

marcmerlin, 2017-06-30:
Please ignore the .zip at the end, this was added to trick github. It's simply an lzma file.
[00000058_flight2_smart_auto_broken_after_landing.BIN.lzma.zip](https://github.com/ArduPilot/ardupilot/files/1116604/00000058_flight2_smart_auto_broken_after_landing.BIN.lzma.zip)

marcmerlin, 2017-06-30:
@tridge @magicrub  I'm not sure if you're tracking plane bugs for the 3.8 release, but between this one, https://github.com/ArduPilot/ardupilot/issues/6510 which may have been a similar issue where the aircraft didn't seem to know which was was up despite the OSD horizon looking ok, and then a flip and a fatal crash in https://github.com/ArduPilot/ardupilot/issues/6522 
I'm kind of wary of 3.8rc5, to the point where I'm thinking I should probably go back to 3.7.1 unless you find obvious things in the logs I've provided.

marcmerlin, 2017-06-30:
I'm not putting the bug label since I don't know for sure that it's a bug, but it smells like one at least.

magicrub, 2017-07-01:
Thanks for the log. I'll take a look.

tridge, 2017-07-12:
hi Marc, I don't see the servo craziness in the log. If you can find it, can you give me the timestamp in the log in terms of seconds since boot? Just graph ATT.TimeUS*1.0e-6 and it will plot seconds since boot.
I only see one FBWA section after the MANUAL 765s section. It is about 1 minute after MANUAL ends, and the servo movements look reasonable to me. I see a strange airspeed increase where I presume you blew into the pitot tube, and some stick input reflected in output, then some surface movement matching attitude change.

marcmerlin, 2017-07-13:
@tridge sorry that I didn't get around to posting the graphs last time.
Don't those 2 graphs show aileron and rudder twiching?
It's as if the plane was sensing that I was quickly tilting it, except I wasn't.
Is it something like an IMU returning bad data?
As a reminder, after a reboot (pulled the battery) everything went back to normal and thankfully this happened after I had finished a flight which also went ok.
[image]
[image]

Naterater, 2019-01-04:
I haven't heard of any problems recently. If this is reproducible, then let's keep this open. Otherwise I would assume that this behavior is not happening to others.

magicrub, 2019-01-04:
I recall there was a change just a couple weeks ago that probably fixed this. I'll try and find out what it was.

IamPete1, 2021-08-15:
Sounds like this has been fixed.

closed 2021-08-15 by IamPete1