ArduPilot/ardupilot #6530: Plane/RFE: allow having airspeed reset by arming

marcmerlin, 2017-07-03:
With multiple airspeed sensors, I end up with bad airspeeds after init. 10kph or more.
Because no existing android GCS lets me reset airspeed like apmplanner does, sometimes I'll power cycle the entire vehicle, but then have to wait for a full re-init and may end up with an airspeed closer to 0, but not always.
Can we please have a non default setting where arming will unconditionally reset airspeed to 0?

I'm adding the safety tag, because routinely taking off with an incorrect airspeed, which usually seems to be positive, tricks the aircraft into thinking it's above stall speed, when it's not, which caused 2 stalls during the flight

marcmerlin, 2017-07-03:
https://youtu.be/wxkHGI85wMU?t=1m31s shows a heavy stall in FBWA with stall avoidance with an airspeed that's supposed to be 50kph (when the stall speed is around 35-40kph).
Stall prevention most likely failed because the airspeed value was wrong.

And again here: https://youtu.be/wxkHGI85wMU?t=20m38s with a measured airspeed around 50kph.

Both times the motor was off and the airspeed sensor is far away on the wing, as far as I can tell the only problem was the 10-13kph offset giving me values that looked safe but were actually too low and caused the airframe to stall.

WickedShell, 2017-07-03:
I'm really confused here, we don't support multiple airspeed sensors yet, so I'm confused how you set up the plane to begin with. Did you just put 2 sensors on the same I2C bus? (This definitely would not go well)

marcmerlin, 2017-07-03:
@WickedShell  The text was unclear "multiple airspeed sensors", I meant one sensor per plane, on different planes, using multiple kinds of airspeed sensors.

WickedShell, 2017-07-03:
@marcmerlin ah, okay, will reevaluate the issue in that context. I got so tripped up on that I was having trouble processing the rest of the information :)

magicrub, 2017-07-04:
@marcmerlin adding a param to trigger an airspeed calibration at arming is certainly easier to do than adding the whole joystick calibration menu thing.. but I am torn between liking this, and considering it a temporary solution until we get the other system online (which has no ETA).

But, as I said. This is easy to implement if we want to do it and is a good idea within it's own scope. However, here comes the hard part: what do we call the param that would enable this?

marcmerlin, 2017-07-04:
@magicrub I'm an incremental "do what can be done now" person. honestly I'm not seeing the other one happening anytime soon, or at all.
Mind you, even after this is done, the calibration mode could still happen later since it does more.
For naming, while I don't love the name, I love consistency more than a non ideal name:
We have ARSPD_SKIP_CAL
So, I'd use ARSPD_ARM_RESET or ARSPD_RESET_ARM

WickedShell, 2017-07-04:
10 kph is 2.77 m/s which is not far outside of the noise floor on the sensors. When you are going 50 kph a ground error of 10 kph is much much smaller. I forget the exact math but last time I looked through this if I was at 57.6 kph, and on the ground had an error of 18 kph, that would only be a 2.52 kph error on reporting when going 57.6 kph.

marcmerlin, 2017-07-04:
@WickedShell I admit to not knowing the noise floor. If there is a noise margin of 2m/s, then I'm going to have to reset my stall speed on my aircrafts (mmh, ok, I have non left right now, but I mean whenever I have new ones) to add a 10kph buffer.
I just don't love this because it forces the autopilot to pitch down longer than it should when it needs to be recovering altitude and I've already had a crash and 2 near once due to the AP not pitching back up when it was supposed to.
Then, and it's tangential, there is the issue that we don't have a stall speed adjusted by motor power. My BFG2600 stalled when the motor was off (gliding) but it can fly at very low airspeed just due to the fact that it has close to 1 to 1 power to weight ratio, which further worsens the scenario of having the aircraft pitch down at full power to gain more airspeed that's totally not needed when in fact it could have already pitched back up and avoided the tree or the ground.

magicrub, 2017-07-04:
I'm in favor of ARSPD_ARM_CAL or ARSPD_CAL_ARM. I don't like "reset" because cal is used universally everywhere else.

WickedShell, 2017-07-04:
@marcmerlin that's the floor on the ground, in flight it would be a lot lower. I'd still recommend keeping a buffer of 1 m/s above your stall speed.

The problem with the reset on arming IMO is that it's not a instant process, it's non blocking and takes ~2 seconds, which makes it a challenge to assess if the arming command is passing or failing.

marcmerlin, 2017-07-04:
@magicrub good point on CAL. Both work IMO.
I'd say whoever would be nice enough to write the feature can indeed decide which one of the 2 is best :)

marcmerlin, 2017-07-04:
@WickedShell at this point, I don't even take off without checking my airspeed with the aircraft armed and the pitot tube blocked. Having arming succeed while the airspeed is still being reset for the next second or two, does not seem like a problem to me.
Obviously this setting would not be on by default, so it can come with appropriate warnings on how to use it and not just arm and throw the plane in the air the very next second.