ArduPilot/ardupilot #6531: 3.8rc5/Plane: wind speed calculation is wrong

marcmerlin, 2017-07-03:
On this airframe, I've often seen totally wrong wind vectors on my OSD like this screenshot:
[image]

lzma -9 of the .bin (not zipped, remove .zip)
[00000010_half.BIN.lzma.zip](https://github.com/ArduPilot/ardupilot/files/1118349/00000010_half.BIN.lzma.zip)

marcmerlin, 2017-07-03:
You can watch around https://youtu.be/wxkHGI85wMU?t=1m29s
It shows at 1:33 61kph alledged wind behind me, and yet my ground speed is only 5kph over my airspeed.
Clearly things don't add up, and by quite a lot.

At https://youtu.be/wxkHGI85wMU?t=47s the wind was 33kph

tridge, 2017-07-12:
hi Marc, 
The EKF was indeed showing high wind:
[image]
The best clue as to why is the GPS heading versus the compass heading:
[image]
that shows about a 47 degree crab angle, which would be consistent with high wind. So either the compass was off by a lot, or you really did have a lot more wind up there than you thought.
Looking at your airspeed versus ground speed at different ground courses gives some clues:
[image]
that seems to show about a 6m/s which, which is 21km/hr. So I think that you either have bad compass calibration or you have bad airspeed sensing.
I think compass error is most likely, particularly given the two compasses don't agree with each other at all.

marcmerlin, 2017-07-12:
@tridge thanks for having had a look.
21kph peak sounds totally reasonable (it was much less on the ground, but sure it's common that it's higher as you gain altitude), but the screenshot I pasted showed 60kph, which was definitely wrong.
So you're saying the 2 compasses didn't agree, even though I had just redone a calibration recently (as required by the upgrade to 3.8).
The compass value I got in the OSD may have been off by a bit but generally looked sane. 
Would you say that compass error would really account for showing 60kph+ wind when it fact the wind was only around 20kph? 
Am I seeing a different value for the compass in the OSD (which again was mostly correct looking) than the one used by AP internally to compute the wind vector?

marcmerlin, 2017-07-12:
More to the point, I noted that around 1:32 in the video, things clearly look wrong.
1) AS 49kph
2) GS 54kph
3) tail wind 61kph
Clearly those things don't even come close to adding up.

[image]

marcmerlin, 2017-08-08:
@magicrub  do you have someone who would have time to look at this? Wind vector calculation really looks wrong. I hope my last 2 updates make it pretty clear? Thanks.

magicrub, 2017-08-08:
@priseborough could possibly confirm compass problems?

priseborough, 2017-08-08:
It looks like a combination of bad compass calibration/interference and variation in wind speed with height. The EKF learned the compass errors, but the wind estimates then took several minutes to recover because the airspeed sensor was being rejected by the EKF for most of the flight. 

The magnetometer had large in-flight biases that had to be learned by the EKF. Both IMU1 and 2 EKF's were consistent.

[image]

There is a significant  airspeed sensor inconsistency that appears at launch and doesn't improve until late in flight. See the airspeed innovations:

[image]

Wind estimates also come good at the same time:

[image]

magicrub, 2017-08-08:
Thanks @priseborough 
I agree that the current airspeed sensors are junk and require constant calibration. Good news @proficnc has a CAN airspeed sensor in the pipeline that should improve things. You would be a good candidate to beta test it if you're willing. Plus, the developer for that sensor is right down the street form you ( @jschall )

magicrub, 2017-08-09:
@marcmerlin you always have the most interesting logs :) Is this ready to be closed?

marcmerlin, 2017-08-10:
@priseborough  thanks for the analysis. The magic happening behind the scenes is definitely not obvious. It looks like in a case like this everything works in the end, but it took a bit over a minute to stabilize.
@magicrub I'm happy to beta test a new wind sensor from @jschall , send me details 
But back to the case at hand, maybe as an RFE, would it make sense for ardupilot not to output/display a wind vector when it doesn't have a high enough confidence level/while it's learning, or output a wind speed of '1' until EKF is working?
Giving bad data is much worse than outputting no data.

magicrub, 2017-08-10:
>>Giving bad data is much worse than outputting no data.

We don't handle the case of having bad data, it is assumed that if it's plugged in then it's working. There is working being done by @WickedShell to detect clogged pitot tubes and this logic could venture into the same realm so its something we'll keep in mind.

The airspeed is pretty much the only sensor that isn't overly fault tolerant. When it gives bad data, bad things happen.

marcmerlin, 2017-08-11:
@magicrub  I was referring to this comment from @priseborough 
"the airspeed sensor was being rejected by the EKF for most of the flight"
If EKF is rejecting the wind sensor, I actually would rather that it shows me a windspeed of -1 or something obvious to say that there is no reliable windspeed data.

As for "The magnetometer had large in-flight biases that had to be learned by the EKF"
Those are theoretically saved and don't have to be relearned for the next flight, correct?

priseborough, 2017-08-11:
The current code does not save mag biases learned by the EKF in flight.

marcmerlin, 2017-08-16:
Thanks @priseborough , so it sounds like there are a few things that can be improved:
1) save mag biases, is there even a good reason not, to?
2) have an option not to show the wind vector until the airspeed sensor is being accepted by EKF
3) have a sanity check to disable airspeed data while it is nonsensical, like the example I gave:
AS 49kph
GS 54kph
tail wind 61kph

marcmerlin, 2017-08-16:
@magicrub , sorry the last comment/questions were meant for you :)

marcmerlin, 2017-11-12:
@magicrub @WickedShell is there something that you'd agree is actionable here, or do you want me to close this? Clearly things can be improved, but it's maybe neither obvious, nor needed by enough people, to matter.

IamPete1, 2021-08-15:
It sounds like this was caused by a hardware/param setup issue, we now have much better support for detecting failing airspeed. `ARSPD_WIND_MAX` and `ARSPD_WIND_WARN`  along with EKF sensor land affinity.

closed 2021-08-15 by IamPete1