ArduPilot/ardupilot #6533: AP/RFE: add an amp threshold for 30%+ throttle that triggers RTL

marcmerlin, 2017-07-03:
Sadly, I just lost my last (and best) aircraft due to an unexplained motor failure in flight.
I reviewed the OSD logs and I had enough altitude to RTL if the AP had immediately turned back after detecting that with more than 30% throttle, amps dropped to below 2A.
Sadly, the airplane was flying around just fine without power, except for losing altitude, and ti took me too long to realize that I had no power. By the time the airplane was heading back, it was too low and I lost all signal at 60 meter altitude and 1.5km distance.
I wasted most of a day trying to look for it on a plotted course from its last recorded course, but sadly the terrain made it not possible.

I chose 30% power as a threshold because on some motors, a lower motor setting may not really spin the motor or use measurable amps.
All all my aircrafts, drawing less than 2A at 30% power is clear proof that the motor is not spinning, but the 2A value should be configurable, and potentially the 30% value too. Leaving them at 0 (default?) would disable the feature.

marcmerlin, 2017-07-03:
Oh, I forgot to mention that while I'll likely never know, the ESC may have detected an unexpected condition and shut down the motor. If I had gotten adequate warning, I could have taken throttle to 0 and back up to reset it, and it's possible that the motor could have restarted, saving the flight and the aircraft.
Ask for ardupilot to try cycling the power down and back up if it detects what it computes as a loss of motor, may be too much, but it could be considered as extra credit.

marcmerlin, 2017-07-11:
@magicrub I saw you happened to see https://www.youtube.com/watch?v=HFFaY7l4X3A
If you watch at 4:35, https://youtu.be/HFFaY7l4X3A?t=4m35s you'll see throttle at 63% and watts dropping from 263 to 23. 
That's clearly where the motor died, but sadly the sound feed was not good enough I didn't hear in my earpiece that the motor just died. I should have noticed that I was losing airspeed and altitude sooner, but by the time I did it was too late to turn back to landing.
The video signal dropped simply because the aircraft got too low. The last video frame I have shows the aircraft in GUIDED mode due to having breached the minimum altitude deck, and trying to fly back. 
I plotted a course of where it may have hit the ground but didn't find it anywhere in that area, no one reported it despite my info being on the aircraft, and more importantly neither of my 2 backup "find me on the ground" devices worked, so I'm now reasonably sure it ended up in the water :(
As I was saying I will work on some OSD code that will report the mismatch between throttle and watts/amps, but ardupilot should/could also detect this and initiate an immediate RTL.

magicrub, 2017-07-11:
this is likely a duplicate of https://github.com/ArduPilot/ardupilot/issues/3456

marcmerlin, 2017-07-11:
Yes, it is, thank you, I'll close this.

OXINARF, 2017-07-12:
@marcmerlin So can it be closed? It is probably good to mention this one in #3456.

closed 2017-07-12 by magicrub

marcmerlin, 2017-07-12:
I could have sworn I clicked the comment and close button, sorry about that.