ArduPilot/ardupilot #6537: Add support for High-G accels

magicrub, 2017-07-04:
Some applications would like to measure high-G (>50G?). Don't necessarily need to navigate using this accel, but would like to at least log it and stream it over mavlink.

WickedShell, 2017-07-04:
Isn't this a hardware issue not anything code wise? (Or at least primarily a hardware one first?)

magicrub, 2017-07-04:
yup. I added the driver label. This is a TODO item.

WickedShell, 2017-07-04:
My point being that it's not a driver issue until someone has a hardware board that we could actually measure this on?

proficnc, 2017-07-04:
Will be using the ICM20649 on the upcoming Cube Red. It's 30G... if someone wants to preempt this and write a driver, that would be appreciated :)

magicrub, 2017-07-04:
@WickedShell I anticipate someone working on this soon, I'll let him chime in when he's ready. He has hardware for an accel that goes up to 400G. Accel only, no gyro. e wants it as a ride-along sensor.

WickedShell, 2017-07-04:
Well never mind, there ya go just drivers.

magicrub, 2017-07-04:
@proficnc great news! See @WickedShell , this is why I made an issue for it.. to start a discussion :)

peterbarker, 2021-07-30:
We support the ICM20649.

There's something else we support with a higher range than that.

I'm closing this as I think a new issue requesting support for specific sensors would make sense - otherwise we've got a "length of string" problem.

closed 2021-07-30 by peterbarker