ArduPilot/ardupilot #6540: MAVLink rangefinder use id number

khancyr, 2017-07-05:
#### Issue details
Currently we update all mavlink rangefinder type each time a distance sensors message arrived. We should only update if the instance number match the id 

Maybe also drop old message ?

Ridhwanluthra, 2018-02-27:
hey i would like to work on this, please tell me what is to be done

peterbarker, 2018-02-27:
@Ridhwanluthra Looks like we need an additional check here: https://github.com/ardupilot/ardupilot/blob/master/libraries/AP_RangeFinder/AP_RangeFinder_MAVLink.cpp#L55 that the incoming message is, indeed, for this backend.

However, @khancyr will have to help here - as he owns this issue ATM!

IamPete1, 2021-08-30:
@khancyr I wonder if https://github.com/ArduPilot/ardupilot/pull/18513 would be sufficient to close this?

IamPete1, 2021-09-07:
@khancyr I think we can close now?

closed 2021-09-07 by khancyr