ArduPilot/ardupilot #6542: ROI Gimbal Pan/Copter Yaw Enhancement

rmonty, 2017-07-05:
#### Issue details
Some 3 axis gimbals do not have full 360 degree panning capability.  When setting up the gimbal and selecting Pan capability, the copter will no longer yaw to try an keep the ROI focused but instead try to Pan the gimbal.  If the gimbal cannot reach the desired angle, sometimes it will behave erratically.  It would be nice if there were still a capability for the copter to yaw to keep the ROI focused knowing the Pan angle limitations of the gimbal.

#### Version
N/A, but using 3.5 currently

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter (but may be relevant to others)
[  ] Plane
[  ] Rover

#### Airframe type
Hexacopter (Tarot 680Pro frame)

#### Hardware type
Pixhawk

#### Logs
N/A

guyht, 2017-07-06:
I would like to see this as well

KirkKirk, 2017-09-16:
Hi Guys,

Recently I got Tarot tl4d (3 axes) and I was trying to make it to work with ROI in mission or "Point camera Here".
In the created circling Mission with ROI in the center of the circle, the camera is pointing to the ROI in the beginning but after the next waypoint it is pointing to random directions but always tilting to the right angle (altitude).
The same situation is with the manual command "Point Camera Here" where I've noticed that the camera yaw is pointing in the wrong direction but to the right altitude. I've followed all instructions and guidance but without success. 
The strange thing is that with manual control (RC ch9 and rc10) it's working properly.
Maybe someone who has this type of gimbal will help me to make it work as expected.

Thanks,
Kirk

amilcarlucas, 2017-10-22:
maybee a parameter can be added, something like "YAW copter if mount PAN is closer to mount PAN limit that this"
When set to 5 deg for instance, it would only change the YAW of the copter if the mount PAN is very close to the mount PAN left or right angle limit. This would then effectively let the mount always have 5 deg of "free" range that could be used for fast movements.

UpsilonAudio, 2018-02-26:
I confirm my interest for this feature as well, the copter’s yaw should change, only if the ROI is going out of the gimbal pan’s range, and should go back to it’s natural value (facing copter’s direction) when the ROI goes back to the gimbal authorized values. Moreover it will be great to stream over telemetry link (MAVLink) the active yaw mode

anbello, 2019-04-02:
I too would like to see this enhancement. Low cost 3 axis gimbal can't do full 360° pan.

Avalonnw, 2019-04-30:
It would be great to have an option to not cotrol the yaw of the gimbal at all; instead still yaw the quad and let the gimbal do its gimbally thing and smooth it out itself.

anbello, 2019-08-05:
I would like to know if this:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Mount/AP_Mount_SToRM32.cpp#L80
means that with a 3D gimbal with Storm32 controllers is always the copter to yaw towards the ROI and not the gimbal (even if it has the capability to do that).