ArduPilot/ardupilot #6545: Baro: extend health check to ensure values are changing

rmackay9, 2017-07-06:
The current AP_Baro health check (https://github.com/ArduPilot/ardupilot/blob/Copter-3.3/libraries/AP_Baro/AP_Baro.cpp#L325) checks:

- data has arrived in the past 0.5 seconds
- pressure is non-zero

We should extend this to also check that the values are changing.  They don't need to change on every read of course but we should have seen some change in pressure or temperature with in the last 0.5 (or more?) seconds.

I'm raising this issue because I've seen one log where the sensor kept responding but provided the same value in every response.

OXINARF, 2017-07-06:
I've seen this request before and, although I agree that is a problem, I'm a bit worried about false positives, specially given most boards only have one baro. What would also be nice was if the EKF behaved with baro the same it does with compass: when it looks like the sensor is giving bad data, change to the next one - this way we wouldn't rely purely on baro library changing the primary.

rmackay9, 2017-07-07:
This is in master now so closing.

closed 2017-07-07 by rmackay9