ArduPilot/ardupilot #6548: Ardupilot + only RPi Zero (W)

karu2003, 2017-07-06:
Hello,

I'm a beginner in the world Ardupilot and I want to start with a simple one.
Help me make configuration for a very simple Ardupilot + RPi Zero (Linux).
I have only RPi Zero. I want to use PPM_in on GPIO4, i have two motors.
I will use PWM (RCOutput) on GPIO12 and GPIO13. 

Thanks for the help.
Andrew.

hiro2233, 2017-07-06:
Hello, try to test with urus board, this one have the input ppm and gpio in/out enabled onlt enabled. Tested with RPi B+, but this should work for RPi Zero. For now it's in development.

karu2003, 2017-07-06:
Thank you,

I found in HAL_Linux_Class.cpp for RCInput. But, I can not find it for RCOutput.

OXINARF, 2017-07-06:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

I suggest you start by reading the developer wiki: http://ardupilot.org/dev/index.html
I have to say that starting by adapting to a new board isn't certainly a simple one.

closed 2017-07-06 by OXINARF

rmackay9, 2017-07-07:
Here is the list of flight controller boards we support. http://ardupilot.org/copter/docs/common-autopilots.html

For RPi2 or RPI3, the NAVIO2 is my personal favourite.