ArduPilot/ardupilot #6555: VTOL-Plane: Face into the wind during QRTL

iskess, 2017-07-08:
#### Issue details
I finally got around to testing 3.8 beta. I needed to make sure this [RTL bug](http://discuss.ardupilot.org/t/quadplane-hybrid-rtl-confusion/14862) I found in 3.7 was fixed, which it is. 
However during these tests I found an issue when I command a Hybrid RTL upwind of Home. The plane turns directly towards home with the wind on its tail, then transitions into QRTL. While this is not an issue for Copter, a Qplane really struggles to hold position with a stiff wind on its tail. Also, it can't benefit from Q_VFWD_GAIN (unless you have reverse thrust set up).  I do have Q_WVANE_GAIN turned on but this does not accomplish the 180 deg turn needed. It actually makes it worse as it slowly tries to turn the plane making it very difficult to control. (shameless plug: #5575 QSimple Mode would help)
I would prefer that the plane remain in forward flight and begin a LOITER turn at RTL_RADIUS until it can approach the land point with a heading into the wind, then transition into QRTL.  I know there are other open issues with similar code, like the request for a helical landing #2279. However it might be sufficient to simply remember the heading used for takeoff and use that for QRTL and QLAND.  

#### Version
3.8b6

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] VTOL-Plane
[  ] Rover

#### Airframe type
3DR Aero (Skywalker)

#### Hardware type
Pixhawk 1

#### Logs
Available if desired.

magicrub, 2017-07-08:
I thought it always did an initial turn-into-wind when a vtol landing was triggered (then correct via vein) . Maybe qrtl isn't trigging that same behavior.

iskess, 2017-07-08:
I only tested the VTOL_LAND command with an approach path heading into the wind. I'm reluctant to test it facing downwind since I nearly crashed yesterday during that QRTL. It got so uncontrollable I had to abort the landing by switching to FBW and punching the throttle. (see QLAND Abort #4446).

This gives me an idea for another possible solution. Allow DO_LAND_START to work with QRTL. Unlike a Copter, a VTOL plane often needs a landing approach depending on its cruise speed and the wind speed.  There may also be obstacles to be avoided and minimal VTOL battery available to accommodate high altitude vertical descents.

iskess, 2017-07-09:
I tested a VTOL_LAND with a cross wind, I wasn't brave enough to use a tailwind again. After transitioning into VTOL is does attempt to orient itself into the wind, but I can't tell if that is just Q_WVANE_GAIN doing it job.  It can still get quite squirrely as its slowly trying to adjust heading while reposition over the landing spot with plenty of wind ( ~6 m/s). 
I still maintain that the smoothest and most reliable method to handle landings with wind is to position the aircraft while in forward flight, before transition occurs.

IamPete1, 2021-07-12:
We can now a VTOL approach, circling and then transitioning into the wind, https://github.com/ArduPilot/ardupilot/pull/16977

closed 2021-07-12 by IamPete1