ArduPilot/ardupilot #6559: RTL in AUTO mission should return at specified height???

RQRobert, 2017-07-09:
#### Issue details
_Please describe the problem, or desired feature_
**RTL in AUTO mission should return at specified height**

#### Version
_What version was the issue encountered with_
N/A

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
quad

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk v1

#### Logs
_Please provide a link to any relevant logs that show the issue_

Description:
As of RTL docs , RTL should climb to RTL_ALT and then return at the same height.
BUT
I use AUTO mission in which altitude for **WP1 is 400mtres** , the the naxt waypoint is return to launch aka **RTL** , the problem is the quadcopter returns home at 400 metres .I want it to return at 10metres .

How could i achieve this OR may be a new feature for developers to achieve ????

RQRobert, 2017-07-09:
For Plane this parameter is 

ALT_HOLD_RTL: RTL altitude
------------------------------

Return to launch target altitude. This is the relative altitude the plane will aim for and loiter at when returning home. If this is negative (usually -1) then the plane will use the current altitude at the time of entering RTL. Note that when transiting to a Rally Point the altitude of the Rally Point is used instead of ALT_HOLD_RTL.

RQRobert, 2017-07-09:
So i hope a similar parameter like ALT_HOLD_RTL for copter should be added.
Otherwise i cant use RTL in AUTO because i have to decent my copter vertically from 400+ metres to HOME which takes a lot of battery amps and i cant have those amps at the end of mission.

magicrub, 2017-07-09:
Implemented as RTL_ALT in centimeters. A value of zero gives you the behavior you're seeing, a nonzero value gives you the behavior you're asking for.
http://ardupilot.org/copter/docs/parameters.html#rtl-alt-rtl-altitude

closed 2017-07-09 by magicrub

OXINARF, 2017-07-09:
@RQRobert I guess that what you missed is that RTL_ALT is the minimum altitude the craft will return home (unless inside the RTL cone, where it can return lower). RTL never goes below the current altitude.

Pedals2Paddles, 2017-07-10:
Copter will not _descend to_ the RTL altitude. It will only climb to it. There aren't any circumstances in copter where the vehicle will descend along the RTL trip home down to the RTL_ALT setting. If it is higher than RTL_ALT, it will climb up RTL_CLIMB_MIN distance only.  If it is lower than RTL_ALT, it will climb up to it.

I can see plenty of use cases for copter to descend along it's trip home.  Especially when you're obscenely high like 400M. However, that becomes very risky business. If the operator has it up high to clear obstacles, descending during RTL would be bad. This is kind of the same reason ArduPilot doesn't automatically retract landing gear. Too many chances for accidents.

Personally I think RTL_ALT should be checked by the operator and set before every flight to verify it is higher than the highest obstacle in the flight area. And RTL crashing into something because it was set too low IMO is operator error. But I think we all know the reality is many many operators don't do that we can't develop something we know will cause people to wreck even if it is technically their fault.

rmackay9, 2017-07-11:
Because this is in a mission, the work around could be to add a command before the RTL command.  This command could be a waypoint near home but with a lower altitude.

Pedals2Paddles, 2017-07-11:
That's exactly what I do. If I want the trip home to be a specific height (or any other specific parameters), I add a waypoint that is right near the home location. Now it will do whatever I want. RTL just handles the final short positioning and landing.

lvale, 2017-07-11:
If we agree this is not an issue, but operation planning I'll close this later today.