ArduPilot/ardupilot #6561: VTOL-Plane: Unexpected FBWA behavior below Q_ASSIST speed

iskess, 2017-07-10:
#### Issue details
The support Wiki says:

> In FBWA mode the quad will assume that pitch stick input is proportional to the climb rate the user wants. So if the user pulls back on the pitch stick the quad motors will try to climb, and if the user pushes forward on the pitch stick the quad motors will try to provide a stable descent.

I was hoping to use this feature to provide an approach descent with some forward speed, but below Q_ASSIST. However when I push the elevator stick forward, the quad motors fight the altitude change. 
In the plot below the yellow line shows me pitching down -20 deg, while the green line shows the quad motors going to max power to maintain altitude. It was not possible to descend using the quad motors in FBWA.

[image]

#### Version
3.8b6

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
QuadPlane

#### Hardware type
Pixhawk1

#### Logs
https://www.dropbox.com/s/o7q8v4uszdc3g0j/QP%20TECS%20Tests.BIN?dl=0

tridge, 2017-08-10:
It did actually put in a negative demanded climb rate, and the plane did descend, but it descended more rapidly than the calculated desired descent rate, so it put on the VTOL motors to slow the descent.
The reason it wanted a slow descent rate is the low throttle. This is the code:
https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/quadplane.cpp#L1078
it multiplies the desired climb rate from the pitch input by the throttle. It assumes low throttle means slow climb or descent.
I'm not really sure what to do about this - the problem is really lack of a way to get the pilots true intent.

iskess, 2017-08-10:
I don't understand why the pilot's throttle input is being considered for climb rate control. Why can it not work just like the wiki says where the throttle stick only controls the horizontal motor in FBWA, and elevator stick controls the throttle of the Qmotors to attain the desired climb rate?
I'm still not seeing the possible confusion over the pilot's intent.  Can you elaborate a bit?
Thank you.