ArduPilot/ardupilot #6564: "Frame unknown" when mavproxy connect to sitl

dongtuu, 2017-07-10:
#### Issue details
I followed the instructions at here http://ardupilot.org/dev/docs/sitl-native-on-windows.html . I run "$ ./ArduCopter.elf --home -35,149,584,270 --model quad" in the cygwin to start the sitl and run "mavproxy.py --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0:14551" in cmd to connect to sitl. But in cmd I got the hint “Frame UNKNOWN”. I test the code simple_goto.py and got the continual hint "waiting for arming". I tried mission planner and came across with "Frame UNKNOWN" too.

#### Version
dronekit (2.9.1)
future (0.16.0)
lxml (3.8.0)
MAVProxy (1.6.1)
monotonic (1.2)
pip (9.0.1)
psutil (5.2.2)
pymavlink (2.2.3)
pyserial (3.3)
setuptools (28.8.0)
six (1.10.0)
uavcan (1.0.0.dev27)

#### Platform
 Copter

#### Logs
C:\Users\Gao>mavproxy --master tcp:127.0.0.1:5760 --sitl 127.0.0.1:5501 --out 127.0.0.1:14550 --out 127.0.0.1:14551
Connect tcp:127.0.0.1:5760 source_system=255
Running script (C:\Users\Gao\AppData\Local\MAVProxy\mavinit.scr)
Running script C:\Users\Gao\AppData\Local\MAVProxy\mavinit.scr
-> set moddebug 2
-> module load help
Loaded module help
Unknown command 'graph timespan 30'
Log Directory:
Telemetry log: mav.tlog
MAV>

Init APM:Copter V3.6-dev (33aec6e8)

Free RAM: 131072
FW Ver: 120
----------------------------------------

load_all took 0us
0 0 0 validate_structures:176: Validating structures
DataFlash_File: buffer size=16384
online system 1
STABILIZE> Mode STABILIZE
APM: PreArm: RC Roll not configured
APM: Calibrating barometer
APM: APM:Copter V3.6-dev (33aec6e8)
APM: Frame: UNKNOWN
Received 855 parameters
Saved 855 parameters to mav.parm
APM: Barometer calibration complete
Init Gyro**no link
link 1 down
*
APM: PreArm: RC Roll not configured

Ready to FLY link 1 OK
fence breach
heartbeat OK
APM: EKF2 IMU0 initial yaw alignment complete
APM: EKF2 IMU1 initial yaw alignment complete
APM: GPS 1: detected as u-blox at 115200 baud
APM: EKF2 IMU0 tilt alignment complete
APM: EKF2 IMU1 tilt alignment complete

rmackay9, 2017-07-10:
I'm surprised the parameter wasn't set but in any case, entering this should help:
param set FRAME_CLASS 1

dongtuu, 2017-07-10:
Hi rmackay9, you really helped me. It had bothered me for a week. I hadn't supposed it to be so easy to be solved.  Thank you very much!

closed 2017-07-10 by dongtuu

OXINARF, 2017-07-10:
> I'm surprised the parameter wasn't set but in any case

That only happens when using *sim_vehicle.py*, when using directly the ArduPilot binary it doesn't go catch a default parameter file.