ArduPilot/ardupilot #6568: FMUv4: Random DSM2 sync loss

VladimirP1, 2017-07-10:
#### Issue details
The PX4Firmware dsm decoder desyncs and gives unreasonable output which triggers failsafe.
When the problems occur _rcs_buf in fmu.cpp is full, that marks some scheduling problems.
To reproduce you should connect DSM2 2048/11ms to pixracer.

#### Version
master
#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[ x] Plane
[  ] Rover
Only plane tested. All platforms may be affected

#### Hardware type
Pixracer V12 from AliExpress with R200 removed

#### Logs
[2017-07-10 13-49-09.tlog.zip](https://github.com/ArduPilot/ardupilot/files/1135232/2017-07-10.13-49-09.tlog.zip)

rmackay9, 2017-07-10:
Very interesting, I think we've seen this issue in Copter-3.5 beta testing as well.
http://discuss.ardupilot.org/t/3-5-rc6-pixracer-dsm-problem/17605

VladimirP1, 2017-07-10:
For me reflashing to 3.4.5 does not help, so I do not think 3.4.6 would work.
But the issue looks similar

VladimirP1, 2017-07-10:
Suspicious line from pref output, does this mean that percise delays in APM are off?
APM_delay: 376440 events, 14 overruns, 2777510979us elapsed, 7378us avg, min 12us max 334288978us 1316080.875us rms

peterbarker, 2017-07-11:
This does differ from  http://discuss.ardupilot.org/t/3-5-rc6-pixracer-dsm-problem/17605 in that 17605 specifically mentions DMSx.

Which receiver is in use here, @VladimirP1 ?

VladimirP1, 2017-07-11:
@peterbarker Original DSM2

VladimirP1, 2017-07-11:
As per http://spektrumrc.com/ProdInfo/Files/Remote%20Receiver%20Interfacing%20Rev%20A.pdf ,
the protocol is the same for DSM2 and DSMX in external mode.
In internal mode yes, the first byte of packet differs between DSM2 and DSMX, but ardupilot DSM driver does not make use of the byte

rmackay9, 2018-03-28:
I think this issue was fixed based on this discussion so closing: https://discuss.ardupilot.org/t/3-5-rc6-pixracer-dsm-problem/17605/9

closed 2018-03-28 by rmackay9