ArduPilot/ardupilot #6583: AC3.5rc8 Need SF40C driver upgrade

roque-canales, 2017-07-12:
Following  #6452

Zero value is to much limited......
It must be better to add as for the RNGFND, min and max value filtered by for example:
PRX_READ_MIN
PRX_READ_MAX.
Like this each one can set it as it wants for improve integration.
and if value is below MIN or above MAX, copter use the last readed value waiting new value entering between MIN and MAX.

rmackay9, 2019-09-17:
Here is a related issue: https://github.com/ArduPilot/ardupilot/issues/8958

peterbarker, 2021-07-30:
Each backend is responsible for reporting distance-min and distance-max in the proximity sensor library.

Do we really need more parameters to override the backend's idea of what its limits are?

rmackay9, 2021-07-30:
@peterbarker, yes, I think we do unfortunately.  It keeps coming up in the forums that people want to limit the minimum range so that close obstacles that are actually on the vehicle are ignored.  The maximum range is also requested sometimes I think.

peterbarker, 2021-07-31:
On Thu, 29 Jul 2021, Randy Mackay wrote:

> @peterbarker, yes, I think we do unfortunately. It keeps coming up in the forums that people want to limit the minimum range so that close obstacles that are actually on the vehicle are ignored. The maximum range is also requested sometimes I think.

The "ignore zones" aren't enough for the short zones?

peterbarker, 2024-07-31:
We have MIN and MAX parameters now in Proximity.

closed 2024-07-31 by peterbarker