ArduPilot/ardupilot #6585: Autotest: Autotest scripts fail due to wrong path on mavproxy

karthikdesai, 2017-07-12:
#### Issue details
Running the autotest.py script with the following parameters. 
This 

`python Tools/autotest/autotest.py -j4 --skip build.All,build.Binaries,build.Examples,build.Parameters,build.ArduPlane,defaults.ArduPlane,fly.ArduPlane,fly.QuadPlane,build.APMrover2,build.ArduCopter,build.Helicopter,fly.CopterAVC,build.AntennaTracker,build.ArduSub,defaults.ArduSub,dive.ArduSub,convertgpx`

This is run on Cygwin64 running on a Win10 PC.

The failure occurs when the script tries to find the path of the telemetery log of the Copter. 

Path of the telemetry log location gets parsed wrongly. This can be seen on both Rover and Copter but only Copter reports this as a failure. The remaining part of the path can be seen 2 lines later

[Rover_Part](https://gist.github.com/karthikdesai/e5602a2852c96777ba1a1baf113b3a4c#file-bamboo_autotest-log-L1239)

[Copter_Part](https://gist.github.com/karthikdesai/e5602a2852c96777ba1a1baf113b3a4c#file-bamboo_autotest-log-L1776)

#### Version
Master

#### Platform
[ * ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type

#### Hardware type
SITL

#### Logs
Log output from the console 
[Logfile](https://gist.github.com/karthikdesai/e5602a2852c96777ba1a1baf113b3a4c)

I can also upload the logs that got generated from the autotest if needed!

OXINARF, 2017-07-12:
@karthikdesai I'll say that although there are some problems with paths, the reason the autotest failrf is because Copter fails to do a mission after hitting the ground: https://gist.github.com/karthikdesai/e5602a2852c96777ba1a1baf113b3a4c#file-bamboo_autotest-log-L3497
Because the autotest is run in sequence and not in individual instances, all following tests fail too.

OXINARF, 2017-07-12:
@peterbarker Can you look at the path issues? Specially https://gist.github.com/karthikdesai/e5602a2852c96777ba1a1baf113b3a4c#file-bamboo_autotest-log-L7936 Thanks.

peterbarker, 2017-07-12:
Please remove `C:\Users\INCAFLOW\AppData\Local/MAVProxy/mavinit.scr` and retry

peterbarker, 2017-07-12:
This line:

https://gist.github.com/karthikdesai/e5602a2852c96777ba1a1baf113b3a4c#file-bamboo_autotest-log-L1239

leads me to suspect MAVProxy.expect() isn't working correctly on Windows.

peterbarker, 2017-07-12:
@karthikdesai Try changing ` `timeout=1` to `timeout=10` on this line: https://github.com/ardupilot/ardupilot/blob/master/Tools/autotest/pysim/util.py#L265

karthikdesai, 2017-07-13:
@OXINARF Wow I didn't even notice the autotest of the copter failing! I was too fixated on getting the directory right!

So the test was successful! The timeout did the trick. 
Thanks @peterbarker . I also removed the mavinit.scr. But the test failed right after

 [Logfile for reference](https://gist.github.com/karthikdesai/448f838a4c3ad624b297ebf370aa94f0)

karthikdesai, 2017-07-13:
The rover still didn't get that right but reports a pass
https://gist.github.com/karthikdesai/448f838a4c3ad624b297ebf370aa94f0#file-bamboo_autotest_success-log-L1190

But copter was no problem
https://gist.github.com/karthikdesai/448f838a4c3ad624b297ebf370aa94f0#file-bamboo_autotest_success-log-L1679

karthikdesai, 2017-07-13:
Right, repeated the same test and it fails with the same error. :/
Perhaps i got lucky the last time but not this time around
[Fail Log](https://gist.github.com/karthikdesai/af499ec2038e23a412b3a4e92bb763e8)

https://gist.github.com/karthikdesai/af499ec2038e23a412b3a4e92bb763e8#file-bamboo_autotest_fail_1-L1678

peterbarker, 2017-07-13:
Is this machine ridiculously loaded or for some other reason would suffer from severe performance problems?

It could theoretically be severe performance issue.

Its more likely there's some sort of buffering going on, and that we need to flush stuff out of mavproxy or something...

karthikdesai, 2017-07-13:
Well throughtout the test, CPU was at 50%-60% ! But yes this server is kinda overloaded

And yeah regarding the flushing part here do you mean?
https://github.com/ArduPilot/MAVProxy/blob/master/MAVProxy/mavproxy.py#L651
added this
`sys.stdout.flush()`
But there were other issues with cygwin with installing mavproxy.  
(Cause i forgot to do a `pip uninstall mavproxy` before running the setup.py scripts! I checked the EGG and still the old mavproxy.py script was there)
But if you think it would solve the issue at hand then will try it!

peterbarker, 2017-07-13:
@karthikdesai Yes, please pursue the flush thing.

karthikdesai, 2017-07-20:
@peterbarker So this got fixed.
`mavproxy.expect('Telemetry log: (\S+)\r\n')` on https://github.com/ArduPilot/ardupilot/blob/master/Tools/autotest/arducopter.py#L1002 did the trick
a PR from @amilcarlucas should be coming up soon

closed 2017-07-20 by karthikdesai