ArduPilot/ardupilot #6587: Could /Tools/autotest/sim_vehicle.sh run commands automatically?

cnpcshangbo, 2017-07-13:
#### Issue details
I am trying to follow this to run the simulation. http://docs.erlerobotics.com/simulation/vehicles/erle_copter/tutorial_5#executing-the-node
It needs me to input commands after running the sim_vehicle.sh script.

```
MAV>param load /[path_to_your_home_directory]/simulation/ardupilot/Tools/Frame_params/Erle-Copter.param
```
Wait until another two command finish in another terminal.
```
MAV>mode GUIDED
MAV>arm throttle
MAV>takeoff 2
MAV>param set SYSID_MYGCS 1
```
Since I need to do this many times to tune controller and marker parameters, is there a method that I can get the simulation started by one command?

#### Version
https://github.com/erlerobot/ardupilot

#### Platform
[  ] All
[  ] AntennaTracker
[X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
Erle-Copter
#### Hardware type
sitl
#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-07-13:
This issues list is for feature requests and confirmed bugs. For development questions please post in the forum (http://discuss.ardupilot.org) or ask in Gitter chat (http://gitter.im/ArduPilot/ardupilot).

Also, we don't obviously support forks of ArduPilot, so if you want to use Erle's fork, you need to ask them for help with issues.
In any case, look at https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/runfliptest.sh and https://github.com/ArduPilot/ardupilot/blob/master/Tools/scripts/runplanetest.py, they are simple examples of what our autotest suite does.

closed 2017-07-13 by OXINARF

cnpcshangbo, 2017-07-20:
Thank you @OXINARF