ArduPilot/ardupilot #6589: Copter: May be a bug in calc_slow_down_distance(...)

luweikxy, 2017-07-13:
site:
https://github.com/ArduPilot/ardupilot/blob/bd2ba1565c41572c32b05d687916f0af1da32a73/libraries/AC_WPNav/AC_WPNav.cpp#L1284

should be changed to:
_slow_down_dist = speed_cms * speed_cms / (2.0f*accel_cmss);

because slow down from v in -a, s = v^2/(**2***a) not  v^2/(**4***a)

luweikxy, 2017-07-14:
It seems that here a = 2*a, same examples are listed as follow:
https://github.com/ArduPilot/ardupilot/blob/master/libraries/AC_WPNav/AC_WPNav.cpp#L639
_limited_speed_xy_cms += 2.0f * _track_accel * dt;
i.e. v = v + 2*a*t. But isn't that should be v = v + a*t?
Why do we define a as 2*a?

amilcarlucas, 2017-10-22:
@luweikxy maybee you can use gitblame in order to find the person that *MUST* answer the questions :)

rmackay9, 2017-11-23:
I'm pretty sure this is done on purpose to ensure the vehicle starts slowing down much earlier than it needs to.  I've ping'd @lthall to see if he agrees.

lthall, 2017-11-23:
@amilcarlucas These are all my fault if there is a problem.

@luweikxy Great catch, thank for looking at the code in such detail.

Stopping acceleration is doubled to account for the leash reducing in length as the aircraft comes to a stop. We should have a comment in there and Randy and I found a place where we should have had it but didn't I think. We will test it.

Thanks.

closed 2018-01-11 by lthall