ArduPilot/ardupilot #6594: copter origin bug with delayed GPS lock

tridge, 2017-07-14:
I have found a serious bug with origin handling with copter which happens when a user doesn't wait for GPS lock before takeoff. The bug can cause either a rapid climb or rapid descent, and could easily crash a vehicle into the ground hard.
I have added a parameter SIM_GPS_LOCKTIME in SITL to make reproducing the bug easy. Set that to 60 to delay GPS lock by 60 seconds. Then takeoff in ALT_HOLD when we don't yet have GPS lock. When we gain GPS lock the copter either climbs or descends rapidly by several meters.
This happens in real vehicles too. I have seen a 4m sudden drop on GPS lock in a real vehicle.
SITL log here:
http://uav.tridgell.net/OriginBug/00000004.BIN

tridge, 2017-07-14:
It turns out that this bug doesn't affect copter-3.5, but does affect master.
I tried running master with the AP_NavEKF2 code from copter-3.5 and the problem goes away, which makes it likely to either be an EK2 bug, or a lack of notification of origin change to the vehicle code.
ping @priseborough

priseborough, 2017-07-15:
The problem is caused by the default value of OGN_HGT_MASK which is currently set to 0. It should be 4 so that changes to reference height that occur when GPS passes checks are applied to the origin and not the local position. However in the interests of making the interface more intuitive I will push a patch which reverses the logic for bit position 2 in that parameter so that a parameter value of 0 results in the expected behaviour.

priseborough, 2017-07-16:
https://github.com/ArduPilot/ardupilot/pull/6610

OXINARF, 2017-07-17:
Fixed with #6610, thanks @priseborough

closed 2017-07-17 by OXINARF