ArduPilot/ardupilot #6595: AP_Proximity: Correct configuration for TeraRanger Tower is not ensured.

danorbtnk, 2017-07-14:
#### Issue details
If the TeraRanger Tower is connected in "text mode" it will not work, i.e. it will be recognized but data is invalid and Mission Planer shows Bad Lidar Health.

The core issue here is, that the tower can be configured for different operating modes and the driver does not ensure, it's in the expected mode of operation. In particular the data transmission can be set to either human readable (text) or binary. The driver implicitly assumes the tower to be operating in the latter.

The tower configuration can be done/happens over UART so it would make sense to ensure the correct operating mode on system startup.

I'm actually currently working on resolving this issue myself, so probably expect a pull request soonish.
 
#### Version
Copter-3.5rc11 

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
should be irrelevant
#### Hardware type
pixhawk, but this should also not matter
#### Logs
don't have any but issue should be perfectly reproducable on the bench and is frankly obvious from looking at the source.

IamPete1, 2021-08-30:
Is this still at issue on 4.1? Seems to have been related to the PX4 hal that we no longer use.

IamPete1, 2021-09-11:
Please re-open if this is still a issue.

closed 2021-09-11 by IamPete1