ArduPilot/ardupilot #6605: Dodeca-hexa ESC wiring angd Propeller Direction

RQRobert, 2017-07-15:
#### Issue details
I am using dodeca-hexa as a hexarotor BUT there are problems with Stability.

ArduCopter v3.5 has added frame_class dodeca-hexa but i m unable to find out how to connect 12 ESCs to Pixhawk 
and what would be there Propeller directions.

#### Version
_What version was the issue encountered with_

#### Platform
[  ] All
[  ] AntennaTracker
[ *** ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_
Dodeca-hexa
#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
Pixhawk v1
#### Logs
_Please provide a link to any relevant logs that show the issue_

OXINARF, 2017-07-16:
@rmackay9 Can you answer this one?

rmackay9, 2017-07-17:
Yes, i'm answering it over here on discourse so I'll close this issue.
https://discuss.ardupilot.org/t/dodeca-hexa-esc-wiriing-and-propeller-direction/19156/4

closed 2017-07-17 by rmackay9