ArduPilot/ardupilot #6606: De-duplicate parameters in Plane

expntly, 2017-07-15:
#### Issue details
I'm seeing more and more parameters that seem like duplicates.  If they're not then the fix could be to just update the doc to reflect how they're different.  Here are the ones I came across recently, there might be more.

RCx_{MIN,MAX,TRIM,REVERSED} and SERVOx_{MIN,MAX,TRIM,REVERSED} parameters
Q_M_PWM_{MIN,MAX} and Q_THR_{MIN,MAX}_PWM

#### Version
Head documentation / 3.8.0-beta5

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

khancyr, 2017-07-15:
Hello, 
RC and SERVO aren't the same : RC is for RC input and SERVO is for output to motors.

I suppose Q param are for quadplane : Q_M_PWM_ should be the range of pwm for motor (us value, mostly 1000 - 2000 see https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Motors/AP_MotorsMulticopter.cpp#L105 )and Q_THR_{MIN,MAX}_PWM the throttle range ( 0 to 100 see https://github.com/ArduPilot/ardupilot/blob/master/ArduPlane/quadplane.cpp#L153)

magicrub, 2017-07-16:
Seem like duplicates is not the same as is duplicate. Each of those are unique. Closing...

closed 2017-07-16 by magicrub

expntly, 2017-07-16:
@khancyr Regarding Q_M params, sorry I'm not seeing a difference in their semantic.  The second one is also a PWM, the range isn't 0 to 100.