ArduPilot/ardupilot #6614: Precision Loiter and Landing: heavy instability when Kalman Filter is enabled

mtbsteve, 2017-07-16:
Copter becomes unstable and bounces in X and Y direction in Precision Landing and Precision Loiter modes as soon as PLND_EST_TYPE: Precision Land Estimator Type is set to 1 (Kalman Filter enabled) 
Unfortunately this is the default mode after installing 3.5. 
Original IR-LOCK sensor connected to I2C2. When I set this parameter to 0 (raw sensor data) everything works well. 

Logs and Video can be found here:
https://discuss.ardupilot.org/t/precision-landing-help-needed/18294/11?u=mtbsteve

#### Version
AC3.5 only 

#### Platform
[  ] All
[  ] AntennaTracker
[ X] Copter
[  ] Plane
[  ] Rover

#### Airframe type
quad

#### Hardware type
Ph2.1

#### Logs
https://discuss.ardupilot.org/t/precision-landing-help-needed/18294/11?u=mtbsteve

fnoop, 2017-07-17:
Possibly related:
#6297 
#5959

mtbsteve, 2019-07-16:
@fnop @oxinarf After 2 years this defect was initially reported, it is still open with Arducopter 3.6.9 along with the IR-Lock sensor for precision landing.
When PLND_EST_TYPE is set to 1, the copter gets out of control and goes into a uncontrollable spiral  when Precision Loiter is enabled and the IR beacon gets in sight. Unlike during Land mode where you can abort the spiraling by switching into another flight mode, disabling precision loiter does not bring back control.
If its not worth fixing, I suggest to amend at least the documentation by ** DO NOT SET PLND_EST_TYPE TO 1 ** when using precision landing with the IR-Lock sensor since you may crash your copter. A log file can be found here: 
https://1drv.ms/u/s!AnKeW8KMoCcyvGIzxmvkuo8Z_yLY

rmackay9, 2019-07-17:
@mtbsteve, I think the kalman-filter version actually works better but the setup is really important.  So there has been some work to allow setting the delay of the sensor which I think was one of the key ways it would go very wrong.

It's sad that we haven't been able to dig into the kalman filter and improve it's robustness.. but hopefully eventually.

mtbsteve, 2019-07-17:
Thanks- but at least there should be a warning added in the wiki until a fix will be available.

peterbarker, 2021-07-30:
@rishabsingh3003 for visibility

IamPete1, 2021-08-30:
@rishabsingh3003 Maybe this has been resolved by your recent work?

khancyr, 2021-08-30:
I am pretty sure it isn't corrected

rmackay9, 2021-08-31:
I think we need to merge this PR from Rishabh to resolve it. We just need to get Rishab's last PR in and then it'll be sorted.

rishabsingh3003, 2021-08-31:
Yes it isn't corrected yet, but hopefully should be once the PR Randy has linked is merged. The issue is a combination of how we are feeding the targets  + direct velocity override to pos controller (from my experiments)

With my PR I was able to precisely land on a target that was shaking around (because the camera could not accurately detect where it was) at 3-4 hz refresh rate with Kalman Filter (and it outperformed the Raw mode everytime)... I am hoping that it will be the only change that is needed because I could not figure out any other problems even with extreme simulations... If not, I am open to tweaking the kalman filter itself

khancyr, 2021-08-31:
I think the instability comes when the data aren't feed at 50hz or with non constant rate. That what I remember from testing and Fnoop feedback

closed 2024-07-31 by peterbarker