ArduPilot/ardupilot #6621: VRBrain build target broken

auturgy, 2017-07-18:
#### Issue details
no waf target for VRBrain boards, make is broken for VRBrain target

#### Version
tried building Master, Copter 3.5, Plane 3.8-dev

#### Platform
[ x ] All
[  ] AntennaTracker
[  ] Copter
[  ] Plane
[  ] Rover

#### Airframe type
_What type of airframe (flying wing, glider, hex, Y6, octa etc)_

#### Hardware type
_What autopilot hardware was used? (pixhawk, pixracer, PX4FMU etc)_
VRBrain (specifically I have vr ubrain5.1, however this effects all targets
#### Logs
_Please provide a link to any relevant logs that show the issue_

ncx94, 2017-07-18:
Nothing has moved on virtual robotix sides since a long time regarding firmware updates...

luca doesn't have enough time to do this unfortunately..

auturgy, 2017-07-18:
So what's the outcome? No longer supported? If that's the case, it would be good to be public about it, so it can be noted in the wiki and missionplanner etc.

OXINARF, 2017-07-18:
Duplicate of #5568

closed 2017-07-18 by OXINARF