ArduPilot/ardupilot #6623: Rover: improve failsafe handling

khancyr, 2017-07-18:
Currently failsafe is very basic, it should be improved to allow to sperate action on different type, loggin etc .

OXINARF, 2017-07-18:
Related to #6208 and #5421

rmackay9, 2018-07-30:
Here's another related issue for the RC failsafe: https://github.com/ArduPilot/ardupilot/issues/9045

jmachuca77, 2018-10-19:
During the China conference, Randy and I had a little chat about Rover failsafes and a couple of things that need to be fixed as they are safety issues

1. When recovering from a GCS failsafe if using RC overrides, the last value of the RC overrides is latched and when recovering the rover starts moving on that direction. This is related to a missing call to:

```c
// clear overrides so that RC control can be regained with radio.
    RC_Channels::clear_overrides();
```
2. The time to detect a GCS failsafe is too long and some hardcoded values. First in:

```c
void Rover::gcs_failsafe_check(void)
```

there is a 2 second hardcoded delay, and second there is a delay dictated by ```fs_timeout``` which currently by default is set to 5 seconds which is too long.

3. Also the heartbeat timer is cleared by writing to ```last_heartbeat_ms```, in copter it is cleared when we receive a heartbeat, and RC_override or a manual_control message. Rover only clears it with the heartbeat which is causing issues with false triggers for us.

Finally because the failsafes are safety related we agreed to mark this as a safety issue.

WickedShell, 2018-10-19:
> When recovering from a GCS failsafe if using RC overrides, the last value of the RC overrides is latched and when recovering the rover starts moving on that direction.

This is only possible if the `RC_OVERRIDE_TIME` hasn't elapsed, otherwise the same effect as calling `clear_overrides` will be achieved. Given that there is a timing parameter for this already I'm not sure that I fully follow the idea of calling `clear_overrides` here. This would trigger new input if you had an RC reciever, but unless I've forgotten anything assuming that `RC_OVERRIDE_TIME` has elapsed (which it will have by default due to the above hard coded delay) this will have no effect.

rmackay9, 2018-10-31:
@jmachuca77, I think the pulsing you're seeing is actually a [MP issue](https://github.com/ArduPilot/MissionPlanner/issues/1972).

rmackay9, 2019-03-13:
I've changed the title on this issue now that we are getting close to resolving them.  I think the last thing we need to do is remove the FS_ACTION parameter and put all the possible values (RTL, Hold, SmartRTL) to the FS_THR_ENABLE and FS_GCS_ENABLE parameters.

We will also need to be sure the parameter upgrade works correctly.  We should check the value in FS_ACTION and if it is non-zero, we should copy it's value to FS_THR_ENABLE (but only if it is also not zero) and FS_GCS_ENABLE (again only if it is not zero)

rmackay9, 2019-03-18:
After thinking about this a bit more I don't think we should remove FS_ACTION because I think users do not actually want the extra complication of different responses for radio vs GCS failsafe.  For example I don't think that a user would want a radio failsafe to trigger RTL, and a GCS failsafe to trigger Hold.  Surely they would want both responses to be the same.

So if there are no objections I think I will close this item.

closed 2019-03-18 by rmackay9

WickedShell, 2019-03-19:
@rmackay9 interestingly I actually do want a difference between my RC and GCS failsafe very very *very* badly on plane. That doesn't force anything with rover of course, but as a user I'd really really like to be able to have different behavior. (Even if the out of box behavior is that one just uses the action from the other so that users only have to think about one unless they want advanced behavior.)

rmackay9, 2019-03-19:
@WickedShell, ok, well.. i can imagine a day when we will consolidate the failsafes across vehicles so maybe at that point we can add the extra option for Rover

peterbarker, 2019-03-19:
On Mon, 18 Mar 2019, Randy Mackay wrote:

> @WickedShell, ok, well.. i can imagine a day when we will consolidate the failsafes across vehicles so maybe at that point we can add the extra option for
> Rover

I actually started to sketch this out a little while ago: 
https://github.com/ArduPilot/ardupilot/compare/master...peterbarker:wip/wind-failsafe?expand=1