ArduPilot/ardupilot #6627: QuadPlane unable to transition

expntly, 2017-07-19:
#### Issue details
Just upgraded to a 3.8.0-beta6 firmware.  I'm trying to transition but the vehicle won't gain enough speed.  Worked OK before.

This is an AUTO mission (VTOL_TAKEOFF, WP, WP ...):

[image]

Not using AirSpeed sensor for this test, flying GPS.

ARSPD_FBW_MIN set to 14m/s.  TRIM_ARSPD_CM 22m/s.

#### Version
3.8.0-beta6

#### Platform
[  ] All
[  ] AntennaTracker
[  ] Copter
[X] Plane
[  ] Rover

#### Airframe type
X8 OctaQuadPlane

#### Hardware type
Pixhawk1

#### Logs
https://www.dropbox.com/s/262i7vys5pob22h/00000025.BIN?dl=0

expntly, 2017-07-19:
Just tested release 3.7.1 and the same thing happened, so I'm removing 3.8.0-beta6 from the title.

There seems to be another conflict somewhere though.  I recently played with Q_WP* parameters but these should be related to strict VTOL navigation only?

2017-07-19 16:55:35.73: PARM {TimeUS : 1304516126, Name : Q_WP_SPEED, Value : 250.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516188, Name : Q_WP_RADIUS, Value : 200.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516247, Name : Q_WP_SPEED_UP, Value : 170.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516413, Name : Q_WP_SPEED_DN, Value : 75.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516473, Name : Q_WP_LOIT_SPEED, Value : 250.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516697, Name : Q_WP_ACCEL, Value : 100.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516761, Name : Q_WP_ACCEL_Z, Value : 100.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304516865, Name : Q_WP_LOIT_JERK, Value : 1000.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304517045, Name : Q_WP_LOIT_MAXA, Value : 100.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304517108, Name : Q_WP_LOIT_MINA, Value : 25.0}
2017-07-19 16:55:35.73: PARM {TimeUS : 1304517194, Name : Q_WP_RFND_USE, Value : 1.0}

expntly, 2017-07-19:
@peterbarker recommended to look at MSG entries in the log to see if the transition starts, and I'm not seeing anything.  I added a D_VTOL_TRANSITION=3 right after VTOL_TAKEOFF just to force the transition, which doesn't seem to help.

2017-07-19 16:55:35.95: MSG {TimeUS : 1304731580, Message : ArduPlane V3.7.1 (22b5c415)}
2017-07-19 16:55:35.95: MSG {TimeUS : 1304731594, Message : PX4: c4dfb9bf NuttX: 258f2257}
2017-07-19 16:55:35.95: MSG {TimeUS : 1304731639, Message : PX4v2 003B0044 34355104 38323831}
2017-07-19 16:55:35.95: MSG {TimeUS : 1304731654, Message : New mission}
2017-07-19 16:55:42.03: MSG {TimeUS : 1310819901, Message : Throttle disarmed}
2017-07-19 16:55:56.35: MSG {TimeUS : 1325138377, Message : Throttle armed}
2017-07-19 16:56:40.23: MSG {TimeUS : 1369013482, Message : Executing nav command ID #84}
2017-07-19 16:57:16.68: MSG {TimeUS : 1405466070, Message : Executing command ID #3000}
2017-07-19 16:57:16.68: MSG {TimeUS : 1405466432, Message : Executing nav command ID #16}
2017-07-19 17:01:04.79: MSG {TimeUS : 1633577263, Message : Throttle disarmed}

tridge, 2017-07-31:
I've reproduced this issue in SITL and I'm working on a fix. Thanks for the bug report!

magicrub, 2017-07-31:
partially fixed by this merged PR: https://github.com/ArduPilot/ardupilot/pull/6700

expntly, 2017-08-13:
I'm taking it this made it into the 3.8.0 release?  If yes I'll close.

tridge, 2017-09-06:
I'd rather this is left over. The problem is reduced in 3.8, but not eliminated completely. It still needs a bit more work.

tridge, 2017-10-21:
I've finally fixed this properly in #7102

OXINARF, 2017-10-22:
#7102 is now merged.

closed 2017-10-22 by OXINARF