ArduPilot/ardupilot #6633: QMC5883L driver causes Cube heading to be off by 90deg

rmackay9, 2017-07-20:
I've found that on my CUBE using a Here GPS with master Copter or Rover, that the vehicle's heading is off by 90degrees (it's turned 90degrees clockwise from it's actual heading).

If I use the COMPASS_TYPEMASK parameter to disable the QMC5883L driver, the heading is correct again after the next reboot.

proficnc, 2017-07-20:
Just to note that neither the cube nor the Here have this chip fitted.

OXINARF, 2017-07-20:
@proficnc Can you say what compasses are in the Here (and The Cube might help too)?

I've found something in the driver that shouldn't be done (writing a register before detection): https://github.com/ArduPilot/ardupilot/blob/master/libraries/AP_Compass/AP_Compass_QMC5883L.cpp#L95 but I'm not sure that's the cause.

rmackay9, 2017-07-20:
It doesn't seem 100% easily repeatable.  I was having this issue constantly with master Rover and Copter but then when I retested again with my own build the problem went away.  Anyway, I will attempt to recreate the issue again tomorrow.

proficnc, 2017-07-20:
Sure.
In a standard first revision here, there is a HMC5983, and an LIS3MDL 
In the Here + an additional MCU9250 as well as the others.

In future Here and here plus units, only a single ICM20948 is fitted due to both the LIS3MDL and the HMC5983 being EOL. 

In the 2.0 cube... (by 3DR) there is a HMC5983 on SPI, and an LSM303D
In the 2.1 cube...  (ProfiCNC/Hex) there are two MPU9250's, and the LSM303D

Pedals2Paddles, 2017-07-20:
Philip, I wonder if this might be why a few of the people who put a here on their solos are randomly unable to get their compass to calibrate despite the settings all being correct?

rmackay9, 2017-07-21:
By the way, I'm pretty sure I've seen this on an mRobotics Pixhawk1 using Plane as well.

OXINARF, 2017-07-21:
@rmackay9 If you reproduce the problem again, please write down what DEVIDs you have.

After a bit of looking at code:

- MPU9250 is likely unaffected since it is on SPI
- same for LSM303D
- LIS3MDL is disabled by default on PH2 (@rmackay9 when you changed the disabled drivers did you have this in consideration? I'm wondering if for some reason you enabled this driver and it is that one causing the problem)
- HMC5983 could be affected, but the register the QMC5883 driver is writing before identification is the first identification register in the HMC; that's only a read register so nothing should happen, but there is always a chance the HMC gets confused with it.

tridge, 2017-07-24:
i suspect its a conflict with an AK8963 on i2c
we need to disable this compass by default I think.
The register the QMC5883 driver is using for its ID is marked as "reversed", not an ID register in the datasheet

tridge, 2017-07-24:
ping @ljwang1102 as author of QMC5883 driver

ljwang1102, 2017-07-24:
I ask the factory engineer,Also can checks the register 0x21,i will fixed it

rmackay9, 2017-07-25:
I've temporarily disabled the QMC compass until we sort out the heading issue, hope that's ok!
https://github.com/ArduPilot/ardupilot/commit/a657daf6355bf3f0d6ff7026897ebd40c0bf0f61

ljwang1102, 2017-07-31:
@rmackay9 Some problems has been repaired, I expect you to test, now this issue only you found

ljwang1102, 2017-08-07:
@rmackay9 I think it should be cancelled #6657,This can't be widely tested,Although I have tested it

rmackay9, 2017-08-14:
I have not seen the 90degree error issue since the changes went in so I'm closing this.

I am seeing issues with the Compass being off by 180degrees now on one of my Rovers which has an early PH2 on it.  I'm still investigating this issue but it seems to be different.

ljwang1102, 2017-08-14:
@rmackay9 I think it might be Calibration problem,You click on pixhawk if you calibrate it,It will set the first compass for the external compass,pixhawk2 and pixhracer have an internal HMC5983,if you not click pixhawk in missionplaner,may be OK

Then I think it should be cancelled #6657,enable_QMC5883,It's a mistake

rmackay9, 2017-08-14:
@ljwang1102, maybe it's a hardware thing but I think it would be best if the QMC could be placed in the same orientation as the HMC by default..

ljwang1102, 2017-08-14:
@rmackay9  QMC have same orientation with HMC,You can not directly modify the GPS circuit board

tdot00cc, 2017-09-12:
I have a Pixhawk 2.1 Cube orientated 180 degrees and 2 Here GPS's orientated 0 degrees.  Both external compasses are enabled for the Here GPS modules and the internal compasses of the cube are disabled.  AHRS orientation is set at 180 yaw, compasses are set at 0.  I'm using a Arducopter 3.5 version.  I'm also seeing the heading being 90 degrees off and 180 degrees off.  Sometimes it works fine and sometimes it is one or the other.  I've done a lot of reboots and one time it fixed by enabling the internal compasses on startup and then disabling them after to remove the inconsistent compasses message.  I have not reproduced this effect 100% of the time though.  I checked the logs and the two external compasses agree but the internal compasses are slightly off most likely because of a carbon fiber frame, however they are disabled.   I will try updating the firmware version to see if it is fixed, as well as the Compass type-mask parameter.

rmackay9, 2017-09-12:
@tdot00cc, yes, I saw it once today as well.  This is very concerning..

proficnc, 2017-09-12:
@tdot00cc  can you plese send me a log?

tdot00cc, 2017-09-12:
Is there an optimal way to fly/test it out so that the log I send you contains the best information?

OXINARF, 2017-09-12:
@rmackay9 Can we change the title of this issue then? 3.5 doesn't have the QMC driver, so it looks like it isn't related.

@tdot00cc The best would probably be to turn on the LOG_DISARMED parameter so that it will be logging when this situation happens.

rmackay9, 2017-09-13:
I've captured logs and @peterbarker and @proficnc and I looked at them today.  The issue, when it occurs, is very obvious if you compare the MAG.MagZ vs MAG2.MagZ.  In cases where the heading is off the two will be reversed (i.e. one positive, one negative) while in the normal situation they're consistent.

I can recreate the issue about 1 in 10 boots and it seems it's easier to recreate if you unplug and then plug back in the battery within a couple of seconds.  So far, if the board is powered down completely (i.e. for 10 seconds) the issue does not occur.

Once the problem occurs, rebooting ardupilot (on the flight controller) does not resolve the problem, only powering down the flight controller resolves it.

We've used mavproxy's devop command to read the individual I2C registers from the compass in bad and good states and this is what we got.  Sorry if the formatting isn't good.

address | location | Name | bad | good
-- | -- | -- | -- | --
0 | 0x00 | Configuration Register A | 00:98 | 00:98
1 | 0x01 | Configuration Register B | 01:20 | 01:20
2 | 0x02 | Mode Register | 02:03 | 02:01 or 02:03
3 | 0x03 | Data Output X MSB Register | 03:01 | 03:00
4 | 0x04 | Data Output X LSB Register | 04:fc | 04:42
5 | 0x05 | Data Output Z MSB Register | 05:fe | 05:01
6 | 0x06 | Data Output Z LSB Register | 06:2d | 06:cd
7 | 0x07 | Data Output Y MSB Register | 07:ff | 07:fe
8 | 0x08 | Data Output Y LSB Register | 08:93 | 08:57
9 | 0x09 | Status Register | 09:01 | 09:01
10 | 0x0A | Identification Register A | 0a:48 | 0a:48
11 | 0x0B | Identification Register B | 0b:34 | 0b:34
12 | 0x0C | Identification Register C | 0c:33 | 0c:33
49 | 0x31 | Temperature Output MSB Register | 31:e1 | 31:08
50 | 0x32 | Temperature Output LSB Register | 32:e0 | 32:40

tdot00cc, 2017-09-13:
I have a log of my crash the other day using dual HERE sensors as external compasses.  Thankfully my drone is a tank and only broke 2 carbon fiber props.  Prior to flight the heading was 90 degrees off but I did the stupid thing and took off anyway because I wanted to get a log of the problem.  The Mags were consistent for the xyz axis until the crash between the 2 Here compasses.  MAGz of the third internal compass was opposite to Magz of the external ones.  Maybe there is a problem somewhere in the initial yaw alignment calculation if there is an orientation difference of the flight controller and external compasses.  I have compass_learn turned of.  If internal compasses are disabled can the autopilot resolve the heading if the AHRS doesn't know which way it is pointing?  I find that if I calibrate internal with external sensors the issue goes away until a couple boots later when compasses become inconsistent, I then disable the internal compass which fixes it until a few boots later where it goes off by 90, 180, 45 degrees.  I see the MagZ difference on some ground tests as well.  I see very good comparisons between the 2 HERE sensors but not so good between the Here sensor and the Cube.  But I have a carbon fibre frame which most likely interferes with the cube.

tdot00cc, 2017-09-13:
Does this issue arise if the Cube and Here sensor are both orientated 0 degrees?

tridge, 2017-09-14:
regarding the HMC5983 rotation error, I have probed the compass on Randys machine, and found that when in the bad state that the -ve self-test bias produces the same output as the +v2 self-test bias. They should produce opposite biases. This may give a way to detect the issue and mark the compass unhealthy.

rmackay9, 2017-09-14:
@tdot00cc, we don't know yet if the orientation of the Cube and Here GPS are important but I suspect it doesn't matter.

It's easy to spot the problem once you know what to look for.  When the Here compass is messed up, the "mz" value (magnetometer Z-axis) is opposite what it normally is.  So If you're in the northern hemisphere the "mz" value should be positive but when the heading is off by 90-degrees, it appears as negative.  if you're in the southern hemisphere the value is normally negative, but when there's a problem it appears as positive.

I live in Japan (which is in the northern hemisphere) so below is a screen capture of the "mz" value when the problem occurred.
[image]

For me, I've only seen this problem if I quickly unplug and plug back in the Cube/Here GPS from a battery.  I've never seen it when using USB power and never when I've left the Cube/HEreGPS for 10 seconds before applying power again.  So I think the short-term, people should wait 10seconds before re-attaching the battery.

tdot00cc, 2017-09-14:
I tried orientating the cube and Here the same way but it doesn't matter.  Also, I am more commonly seeing the heading off 180 degrees not 90.  But the same thing is happening with mz, one being positive and one being negative.  I'm also seeing that sometimes when the issue happens it's not just mz but the my and mx values are inverted as well.  I'm seeing it on usb power as well and 10 seconds seems to resolve the issue.

amilcarlucas, 2017-09-21:
I had this issue today. Using PH2.1, Here as GPS1, AsteRX-m2 as GPS2 and arducopter from master.
It took off fine with all 3 Mag.z positive. Then 3 minutes into the flight it decided to erratically rotate 90° whenever it reached a waypoint :( 
Not nice when you have a 2D gimbal.... yaw is VERY messed up

tdot00cc, 2017-09-25:
I noticed the yaw rotation 90 degrees inflight as well when the ROI was the center of a circular flight path and when it initially flew fine.

amilcarlucas, 2017-09-26:
@karthikdesai and I have a second copter with the same GPS hardware (GPS1 : uBlox and GPS2 : SBF AsteRX-m2) and running the same FW, but on a pixhawk1. We also use the same missions and parameters on both copters.
PH1 flies fine and never had this issue. PH2.1 has it around 60% of the time.

amilcarlucas, 2017-09-27:
Can this be hardware error that only pops in some boards ?
@proficnc can you look at the logs I provided ?
Can you send me a PH2.1 so that I can test on a second board ? I already invested two weeks on this. And so far the only fix I have found is to use a PH1

proficnc, 2017-09-27:
I don't see your specific log?

amilcarlucas, 2017-09-27:
It's the one from Karthik, we work at the same company.

proficnc, 2017-09-28:
Have you switched GPS's between the two platforms?
If the issue is a hardware issue with the Here gps, then it should follow from the PH2, to the Ph1

amilcarlucas, 2017-09-28:
No, I have not.
I tested with the Here on the PH2.1 and the problem is there.
Then I replaced it with the same 3DR ublox GPS that was supplied with the PH1 and the problem persists.

So I think it it the PH2.1 itself that is causing the problem.

Copter1: PH1    GPS1 : 3DR uBlox and GPS2 : SBF AsteRX-m2
Copter2: PH2.1 GPS1 : 3DR uBlox and GPS2 : SBF AsteRX-m2

Copter nr.2 has issues regardless if I use 3DR uBlox or here GPS

proficnc, 2017-09-28:
I can't think of any hardware issues that could possibly cause one part of a compass i2c data stream to change... 
the 2.1 does however have more going on internally than the PH1

I think we may need to get Paul looking at this..

amilcarlucas, 2017-09-28:
Well I disabled both internal compasses.

tdot00cc, 2017-09-28:
When I made the switch I changed from PH1 & 3DR uBlox to PH2.1 & Here & 3DR ublox.  The problem arose after the switch so the problem could be related to the Here GPS or the Pixhawk2.1.

amilcarlucas, 2017-09-28:
My boss asked me to buy PH1s instead because of this. So I might have to abandon my PH2.1.

tdot00cc, 2017-09-28:
Has the problem occurred on earlier firmware versions than 3.5.2?

lvale, 2017-09-28:
@amilcarlucas check your pm on gitter

amilcarlucas, 2017-09-28:
I did not test it. Been running master.
I could try to go back to 3.4.3 or something like that. But that is not a solution for me, I would have to change a lot of my code into that codebase and that would cost me a ton of time.

tdot00cc, 2017-09-28:
I should have my copter up and running again the next few days.  I'm going to try the PH2.1 and 3dr uBlox combo to see if I reproduce the problem.

peterbarker, 2017-09-28:
@tdot00cc Can you provide logs of the flight where it seemed to gain a 90 degree yaw error in-flight, please?

amilcarlucas, 2017-09-28:
On the log from Karthik it happened at second 535 aprox.

amilcarlucas, 2017-09-28:
I bought  a APM 2.6 stand alone compass and could try with that next week.

tdot00cc, 2017-09-28:
I will see if I can dig that one up.  I'm not sure I saved that one because at that point I wasn't trying to figure out/resolve the problem.

rmackay9, 2017-09-28:
@amilcarlucas, @peterbarker and I had a chat about his problem today and @karthikdesai's log shows his vehicle's issue is quite different.  In Karthik's log the compass values are fine, the vehicle's estimated heading is also fine.  It's more than there's some yaw control funniness.  I'm not saying @karthikdesai's vehicle's issue is not a problem, I'm just saying that it's quite different and we shouldn't muddy the waters by including it in this discussion.

JacksonUAS, 2017-10-04:
Currently have this issue on 2 Aircraft, Hex + with Here+ and Here - 3.5.3, and Quad X with Here - 3.5. Both PH2.1 We have it happening with the Hex more often than being correct, so if you need some test data - I have the machine to use! Have been going mad trying to troubleshoot it. Have both 90 and 180 degree out issues on boot, and once had it spinning in 360 circles.

JacksonUAS, 2017-10-04:
Trouble shooting -

Known good compass param copied from (apparently) identical airframe to misbehaving one - Issue still exists.

Misbehaving airframe rolled back to 3.4.6, flashed with above compass param - Issue still exists.

Flashed to 3.6 Dev Arducopter, flashed with above compass param - Issue still exists.

Are we sure this isnt something to do with the new onboard calibration?

lvale, 2017-10-05:
No similar issues seen when using onboard calibration on PixRacer or PixHawk

JacksonUAS, 2017-10-06:
Bunch of testing, looks to be a hardware issue.

Pedals2Paddles, 2017-12-06:
Any news or resolution on this?

JacksonUAS, 2017-12-06:
Hardware issue on Heres and Here+, transient voltage remains after being powered down, which results in the magnetometer booting up in bizarre ways. Adding a resistor to the test pads to discharge the circuit solved the issue.

proficnc, 2017-12-07:
Correct, only occurs after a brownout. 
Adding a 1K resistor across 5v and ground removes the issue.

This only affects people doing rapid power cycling. 

Discussion continues on detecting this sort of issue and causing a prearranged failure based on this.

Issue affects all HMC5983 compasses that don’t have a chance to restart cleanly, not just here.

Pedals2Paddles, 2017-12-07:
Interesting.  How much time should lapse when rebooting to avoid this?

proficnc, 2017-12-07:
30s is good

rmackay9, 2017-12-07:
For my Rover that sees this issue all the time, 10seconds is more than enough.  Below 6 seconds is when I see problems.

rmackay9, 2017-12-07:
Re Here vs other brands of GPS/Compass modules using the HMC... I've only ever seen this happen with the Here GPS so I hate to disagree but I really think that there's a difference with the Here hardware that makes the brownout much much more likely.

[43 more replies]

closed 2018-06-25 by WickedShell